FreeOpcUa/opcua-asyncio #10: background tasks and subcriptions

oroulet, 2019-02-17:
@cbergmiller  In fact I was trying to debug exactly that stuff right now. but I am not really sure how to do this correctly. I see that you implemented the connection renewal using create_task and recursion. This kind of fire and forget stuff in asyncio is asking for trouble. I believe, but correct me if I am wrong, that we should have a a background task, keep reference to task, and at the end cancel it and await it. This is easy to do for connectino renewale but much more difficult for subscriptions

cbergmiller, 2019-02-17:
I agree that the "fire and forget" style can be problematic. If the canceling of the background tasks works and allows the event loop to be shut down in a timely fashion then i have no reservations against using them.

oroulet, 2019-02-17:
Ok i can write a PR later today

-------- Original Message --------
On Feb 17, 2019, 12:04, cbergmiller wrote:

-------- Original Message --------

closed 2019-02-24 by oroulet