FreeOpcUa/opcua-asyncio #18: Can't connect to Rockwell OPCUA server

Ilyazyk, 2019-02-24:
Hello,
opcua-asyncio version stops with error on connection to server step in ` async with Client(url=url) as client`. At the same time opcua works well.

`import asyncio
import sys
sys.path.insert(0, "..")
import logging
from asyncua import Client, Node, ua

logging.basicConfig(level=logging.DEBUG)
_logger = logging.getLogger('asyncua')

async def main():
    url = 'opc.tcp://localhost:4990/FactoryTalkLinxGateway1'
    try:
        async with Client(url=url) as client:
            root = client.get_root_node()
            _logger.info('Objects node is: %r', root)
    except Exception:
        _logger.exception('error')
if __name__ == '__main__':
    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(main())
    loop.close()`

Debug info:
INFO:asyncua.client.client:connect
INFO:asyncua.client.ua_client.UaClient:opening connection
DEBUG:asyncio:Get address info localhost:4990, type=<SocketKind.SOCK_STREAM: 1>
DEBUG:asyncio:Getting address info localhost:4990, type=<SocketKind.SOCK_STREAM: 1> took 0.000ms: [(<AddressFamily.AF_INET6: 23>, <SocketKind.SOCK_STREAM: 1>, 0, '', ('::1', 4990, 0, 0)), (<AddressFamily.AF_INET: 2>, <SocketKind.SOCK_STREAM: 1>, 0, '', ('127.0.0.1', 4990))]
DEBUG:asyncio:poll took 0.000 ms: 1 events
DEBUG:asyncio:connect <socket.socket fd=256, family=AddressFamily.AF_INET6, type=SocketKind.SOCK_STREAM, proto=0> to ('::1', 4990, 0, 0)
DEBUG:asyncio:poll took 0.000 ms: 1 events
DEBUG:asyncio:<socket.socket fd=256, family=AddressFamily.AF_INET6, type=SocketKind.SOCK_STREAM, proto=0, laddr=('::1', 50124, 0, 0), raddr=('::1', 4990, 0, 0)> connected to localhost:4990: (<_SelectorSocketTransport fd=256 read=polling write=<idle, bufsize=0>>, <asyncua.client.ua_client.UASocketProtocol object at 0x037A7FF0>)
DEBUG:asyncio:poll 4000.000 ms took 0.000 ms: 1 events
INFO:asyncua.client.ua_client.UASocketProtocol:open_secure_channel
DEBUG:asyncua.client.ua_client.UASocketProtocol:Sending: OpenSecureChannelRequest(TypeId:FourByteNodeId(i=446), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeId(i=0), Timestamp:2019-02-24 08:55:37.973007, RequestHandle:1, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:OpenSecureChannelParameters(ClientProtocolVersion:0, RequestType:0, SecurityMode:1, ClientNonce:b'', RequestedLifetime:3600000))
DEBUG:asyncio:poll 4000.000 ms took 0.000 ms: 1 events
INFO:asyncua.client.ua_client.UaClient:create_session
DEBUG:asyncua.client.ua_client.UASocketProtocol:Sending: CreateSessionRequest(TypeId:FourByteNodeId(i=461), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeId(i=0), Timestamp:2019-02-24 08:55:37.975007, RequestHandle:2, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:10, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:CreateSessionParameters(ClientDescription:ApplicationDescription(ApplicationUri:urn:freeopcua:client, ProductUri:urn:freeopcua.github.io:client, ApplicationName:LocalizedText(Encoding:2, Locale:None, Text:Pure Python Async. Client), ApplicationType:1, GatewayServerUri:None, DiscoveryProfileUri:None, DiscoveryUrls:[]), ServerUri:None, EndpointUrl:opc.tcp://localhost:4990/FactoryTalkLinxGateway1, SessionName:Pure Python Async. Client Session1, ClientNonce:b'\xe4IK\x81E\x10\xdf&\xff\xdaY\xc5\x9cc\xa1\xa6\xc0\xff\xbd\xcbN\xff\x91\x12J\xacJ\xbe\x9b\xba\xe7\x95', ClientCertificate:None, RequestedSessionTimeout:3600000, MaxResponseMessageSize:0))
DEBUG:asyncio:poll 10000.000 ms took 0.000 ms: 1 events

INFO:asyncua.client.client:find_endpoint [EndpointDescription(EndpointUrl:opc.tcp://localhost:4990/FactoryTalkLinxGateway1, Server:ApplicationDescription(ApplicationUri:urn:TEMP09-81OV7KU9:FactoryTalk Linx Gateway OPC UA Server, ProductUri:urn:FactoryTalk Linx Gateway OPC UA Server, ApplicationName:LocalizedText(Encoding:3, Locale:en, Text:FactoryTalkLinxGateway), ApplicationType:0, GatewayServerUri:None, DiscoveryProfileUri:None, DiscoveryUrls:['opc.tcp://localhost:4990/FactoryTalkLinxGateway1']), ServerCertificate:[...cuted...] , SecurityMode:1, SecurityPolicyUri:http://opcfoundation.org/UA/SecurityPolicy#None, UserIdentityTokens:[UserTokenPolicy(PolicyId:Anonymous_Policy, TokenType:0, IssuedTokenType:None, IssuerEndpointUrl:None, SecurityPolicyUri:None)], TransportProfileUri:http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary, SecurityLevel:11)] <MessageSecurityMode.None_: 1> 'http://opcfoundation.org/UA/SecurityPolicy#None'
INFO:asyncua.client.ua_client.UaClient:activate_session
DEBUG:asyncua.client.ua_client.UASocketProtocol:Sending: ActivateSessionRequest(TypeId:FourByteNodeId(i=467), RequestHeader:RequestHeader(AuthenticationToken:ByteStringNodeId(b=b'\x1an]\xd6[\x04&\x95!w\xb5\x98\xb53\xa0g2\\A<\xc1\xd8\xb9\x10\x9e\x0b\xdf\x10\xea\x00,j'), Timestamp:2019-02-24 08:55:37.979007, RequestHandle:3, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:10, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:ActivateSessionParameters(ClientSignature:SignatureData(Algorithm:http://www.w3.org/2000/09/xmldsig#rsa-sha1, Signature:b''), ClientSoftwareCertificates:[], LocaleIds:['en'], UserIdentityToken:AnonymousIdentityToken(PolicyId:Anonymous_Policy), UserTokenSignature:SignatureData(Algorithm:None, Signature:None)))
DEBUG:asyncio:poll 10000.000 ms took 78.000 ms: 1 events
WARNING:asyncua.client.ua_client.UASocketProtocol:ServiceFault from server received  in response to ActivateSessionRequest
ERROR:asyncua:error
Traceback (most recent call last):
  File "C:\Users\Labuser\git\opcua-asyncio\examples\client-minimal_my.py", line 14, in main
    async with Client(url=url) as client:
  File "..\asyncua\client\client.py", line 65, in __aenter__
    await self.connect()
  File "..\asyncua\client\client.py", line 209, in connect
    await self.activate_session(username=self._username, password=self._password, certificate=self.user_certificate)
  File "..\asyncua\client\client.py", line 404, in activate_session
    return await self.uaclient.activate_session(params)
  File "..\asyncua\client\ua_client.py", line 272, in activate_session
    data = await self.protocol.send_request(request)
  File "..\asyncua\client\ua_client.py", line 124, in send_request
    self.check_answer(data, " in response to " + request.__class__.__name__)
  File "..\asyncua\client\ua_client.py", line 133, in check_answer
    hdr.ServiceResult.check()
  File "..\asyncua\ua\uatypes.py", line 224, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadTimeout: The operation timed out.(BadTimeout)
DEBUG:asyncio:Close <_WindowsSelectorEventLoop running=False closed=False debug=True>
ERROR:asyncio:Task was destroyed but it is pending!
source_traceback: Object created at (most recent call last):
  File "C:\Users\Labuser\git\opcua-asyncio\examples\client-minimal_my.py", line 24, in <module>
    loop.run_until_complete(main())
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python37-32\lib\asyncio\base_events.py", line 571, in run_until_complete
    self.run_forever()
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python37-32\lib\asyncio\base_events.py", line 539, in run_forever
    self._run_once()
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python37-32\lib\asyncio\base_events.py", line 1767, in _run_once
    handle._run()
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python37-32\lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\Labuser\git\opcua-asyncio\examples\client-minimal_my.py", line 14, in main
    async with Client(url=url) as client:
  File "..\asyncua\client\client.py", line 65, in __aenter__
    await self.connect()
  File "..\asyncua\client\client.py", line 204, in connect
    await self.create_session()
  File "..\asyncua\client\client.py", line 340, in create_session
    self._renew_channel_task = self.loop.create_task(self._renew_channel_loop())
task: <Task pending coro=<Client._renew_channel_loop() running at ..\asyncua\client\client.py:353> wait_for=<Future pending cb=[<TaskWakeupMethWrapper object at 0x0383E710>()] created at C:\Users\Labuser\AppData\Local\Programs\Python\Python37-32\lib\asyncio\base_events.py:396> created at ..\asyncua\client\client.py:340>

oroulet, 2019-02-24:
Looks like some issues with that specific server. Can you record a session with Wireshark and attach the log? Do you use encryption? If so disable it

Ilyazyk, 2019-02-24:
Here is a log. Not sure It catched everything (I never used it before). The encryption is set to None.
[wireshark_log.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/2898166/wireshark_log.zip)
And thank you for your work!

Ilyazyk, 2019-02-24:
I gave it more time to log...
[wireshark_log2.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/2898197/wireshark_log2.zip)

elka0711, 2019-02-25:
Same Problem with PQM-1588 OPC-UA.
Do you need capture recording of the failed session?

oroulet, 2019-02-25:
@Ilyazyk both recording contain almost nothing.... maybe everythong is on a local windows PC? then you have an issue and must google for using wireshark on windows

Ilyazyk, 2019-02-26:
@oroulet please try new log. This time I enabled all protocols. Hope there something in.
[wireshark_log.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/2904077/wireshark_log.zip)

Ilyazyk, 2019-03-02:
@oroulet,  I couldn't manage logging (nothing with opcua protocol logged in), but the code from python-opcua works fine:
`import sys
sys.path.insert(0, "..")
from opcua import Client
url = 'opc.tcp://localhost:4990/FactoryTalkLinxGateway1'
if __name__ == "__main__":
    client = Client(url)
    try:
        client.connect()
        root = client.get_root_node()
        print("Children of root are: ", root.get_children())
    finally:
        client.disconnect()
`
And sync version of opcua-acyncio doesn't:
`import sys
sys.path.insert(0, "../..")
from asyncua.sync import Client, start_thread_loop, stop_thread_loop
if __name__ == "__main__":
    start_thread_loop()
    url = 'opc.tcp://localhost:4990/FactoryTalkLinxGateway1'
    client = Client(url)
    try:
        client.connect()
        print("Children of root are: ", client.nodes.root.get_children())
    finally:
        client.disconnect()
        stop_thread_loop()`

with errors:
ServiceFault from server received  in response to ActivateSessionRequest
Traceback (most recent call last):
  File "C:\Users\Labuser\git\opcua-asyncio\examples\sync\client-minimal.py", line 10, in <module>
    client.connect()
  File "../..\asyncua\sync.py", line 91, in wrapper
    result = _tloop.post(aio_func(*args, **kwargs))
  File "../..\asyncua\sync.py", line 40, in post
    return futur.result()
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python37-32\lib\concurrent\futures\_base.py", line 432, in result
    return self.__get_result()
  File "C:\Users\Labuser\AppData\Local\Programs\Python\Python37-32\lib\concurrent\futures\_base.py", line 384, in __get_result
    raise self._exception
  File "../..\asyncua\client\client.py", line 208, in connect
    await self.activate_session(username=self._username, password=self._password, certificate=self.user_certificate)
  File "../..\asyncua\client\client.py", line 403, in activate_session
    return await self.uaclient.activate_session(params)
  File "../..\asyncua\client\ua_client.py", line 272, in activate_session
    data = await self.protocol.send_request(request)
  File "../..\asyncua\client\ua_client.py", line 124, in send_request
    self.check_answer(data, " in response to " + request.__class__.__name__)
  File "../..\asyncua\client\ua_client.py", line 133, in check_answer
    hdr.ServiceResult.check()
  File "../..\asyncua\ua\uatypes.py", line 224, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadTimeout: The operation timed out.(BadTimeout)

oroulet, 2019-03-02:
Thanks. That is interesting. We need to find the difference...hasn't Rockwell a sample server we could try? Otherwise you manage to dump to record one with python-opcua and one with asyncio

Ilyazyk, 2019-03-31:
@oroulet, how can I send you a link to download software?

Ilyazyk, 2019-05-02:
I've found the issue, but could not commite the change. Future's timeout (in Sec) is taken from Client, and for Request the default value (in mSec) has been changed from 10 to 1000.
In class UASocketProtocol:

    async def send_request(self, request, timeout=1000, message_type=ua.MessageType.SecureMessage):
        """
        Send a request to the server.
        Timeout is the timeout written in ua header.
        Returns response object if no callback is provided.
        """
        data = await asyncio.wait_for(
            self._send_request(request, timeout, message_type),
            self.timeout if self.timeout else None
        )
        self.check_answer(data, f" in response to {request.__class__.__name__}")
        return data

cbergmiller, 2019-05-02:
@Ilyazyk thanks for pointing this out! This is a pretty severe problem. I will fix this as ASAP.

oroulet, 2019-05-02:
I do not understand, where is the issue?

Ilyazyk, 2019-05-02:
@oroulet, timeout with default value 10 was used for both: request and future. For request it was not enought on systems with a lot of nodes. That is why we had a BadTimeout.

cbergmiller, 2019-05-02:
`UASocketProtocol._send_request()` expects `timeout` param to be in milliseconds. `UASocketProtocol.send_request()` expects it in seconds and passes the default timeout param of 10 seconds along to `_send_request` without converting it to in milliseconds.
We should refactor our methods to the same unit for timeouts everywhere.

PR #46 (2019-05-03): Fix for issue #18
I changed the unit timeout parameter of all client methods to seconds.
Fixes #18
closed 2019-05-03 by cbergmiller