FreeOpcUa/opcua-asyncio #20: missing tests from python-opcua

oroulet, 2019-02-25:
@cbergmiller do you have any overview of what tests are missing in asyncua compared to python-opcua? Now that I have sync wrapper that seem to completely working, it should in theary be possible to just copy the missing files to this repository, although it would be better to port them to pytest too

cbergmiller, 2019-02-28:
Sorry for the late answer. I can't remember at waht point i stopped when i was porting the tests to pytest. I may have time over the weekend to look into that.

oroulet, 2019-03-03:
@cbergmiller I tried haivng a look but could not find any not portes tests (apart from tools, but these tests are bad, I want to refactor them

swamper123, 2020-12-15:
Are there still some tests missing @cbergmiller @oroulet (it is a longer while ago)?
Otherwise I would add this issue to the ToDo List or close it.

oroulet, 2020-12-15:
I think we close that one

closed 2020-12-15 by oroulet