FreeOpcUa/opcua-asyncio #21: is returning a coroutine from sync methods really more efficient flagging that method async?

oroulet, 2019-02-25:
@cbergmiller is returning a coroutine from sync methods really more efficient flagging that method async?

I see that you often do not declare methods returning coroutine with async....

Is really

```
async def toto():
    await asyncio.sleep(1)

def whaterver():
    return toto()

await whatever
```

faster than 

```
async def toto():
    await asyncio.sleep(1)

async def whaterver():
    return toto()

await whatever()
```
?

I am expecting python to create exactly ONE task when calling whatever() independantly if it is flagged with async or not. But the first version is much less readable since we do not have async declaration

oroulet, 2019-02-25:
If you think it is, I would really like a pointer to the reason for that

cbergmiller, 2019-02-26:
I think your second example should be:
```python
async def toto():
    await asyncio.sleep(1)

async def whaterver():
    return await toto()

await whatever()
```
In the second case two tasks have to be created by the event loop. Creation of tasks is more expensive than a plain function call. I did some measurements of impact for a simple case. If i can find the code i will post it here. Would only be an issue for tasks that are called frequently. I may have been nitpicking on this issue. Premature perfomance tunin g is considered to be an anti-pattern by some.

There is a `Coroutine` type hint available to make the first case type-save though:
```python
async def toto():
    await asyncio.sleep(1)

def whaterver() -> Coroutine:
    return toto()

await whatever
```

oroulet, 2019-02-28:
OK I ran

```Python
import asyncio
import cProfile

async def toto():
    return await asyncio.sleep(0.00001)

async def whatever():
    return await toto()

async def main():
    for _ in range(100000):
        await whatever()

cProfile.run('asyncio.run(main())')
```

against 

```Python
import asyncio
import cProfile

async def toto():
    return await asyncio.sleep(0.00001)

def whatever():
    return toto()

async def main():
    for _ in range(100000):
        await whatever()

cProfile.run('asyncio.run(main())')
```

and the result is:
python3.7 t.py 
         6500298 function calls (6500296 primitive calls) in 4.149 seconds

python3.7 s.py 
         6400298 function calls (6400296 primitive calls) in 4.227 seconds

I admit it seems to be a few more function calls but the performance difference does not seem to be very large......
Unless you find an error in test above I propose to declare async everything that is async for readability. In addition this is the kind of things that could be optimised low level with jit or whatever

cbergmiller, 2019-02-28:
Thanks for checking. I agree with you!

closed 2019-03-03 by oroulet