FreeOpcUa/opcua-asyncio #22: BadTimeout Error connecting to Frako PQM-1588 OPCUA server

elka0711, 2019-02-26:
`opcua.ua.uaerrors._auto.BadTimeout: The operation timed out.(BadTimeout)`
For a better understandig, with the asyncronious client I get that timeout error. See attachment of the output.dump file
[output.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/2904702/output.zip)

The client IP is 10.2.202.253, the Server IP is 10.2.202.200.
Authentification method is anonymous, no  encryption, no username&password

Exception:
`File "/python-opcua/opcua/common/node.py", line 109, in get_data_type_as_variant_type
result = await self.get_attribute(ua.AttributeIds.DataType)`
 `File "/python-opcua/opcua/common/node.py", line 286, in get_attribute
result = await self.server.read(params)`
`File "/python-opcua/opcua/client/ua_client.py", line 319, in read
data = await self.protocol.send_request(request)`
`File "/python-opcua/opcua/client/ua_client.py", line 121, in send_request
await asyncio.wait_for(future, timeout if timeout else None)`
`File "/tasks.py", line 362, in wait_for
concurrent.futures._base.TimeoutError`

If testing a connection with the software UaExpert V1.4.4 to the same OPC-UA Server everything is good

oroulet, 2019-02-26:
you will need to record a session with wireshark both a working one (with uaexpert) and a not working one from python and compare them to try to catch any difference...

elka0711, 2019-02-27:
Ok, I will try it with wireshark, but it is not so easy, because it's a customer system and I have not full admin rights on their system.
 That output.dump file I generate with tcpdump from our gateway at the customer side.
I'll keep you posted. 
Another  try with your client_minimal.py example file was successfull. Not every time, but sometimes. So I think, may be, it is also a problem with the OPC-UA Server implementation on the PQM device.

oroulet, 2019-02-27:
I thing tcpdump can dump in a Wireshark friendly format. But you need a session from uaexpert to compare

elka0711, 2019-02-27:
I know, but I need the admin rights on their system to install and use wireshark for capture the UaExpert.

In my output.dump file I found out that everything runs good as far as time 0.833268 and the timeout comes 10 seconds later at time 10.35665. 
The reason seems to be a UA Secure Converastion problem.
[image]

oroulet, 2019-02-27:
OK. I did not see you attached the tcpdump already. That stuff looks strange. It is a simple read request and the server does not answer.... Can it be some network issue?
I think if it was a session timeout we would get an error messgae from server. Here it just stops sending things. looks like a network or server issue.... it looks like you will need to contact the supplier

oroulet, 2019-02-27:
you can still record an uaexpert session and try to stop a difference if uaexpert works

elka0711, 2019-02-27:
I will contact the supplier, because we can exclude a network issue.  
Also interesting is that UaExpert just the same have problem's every minute.
[image]

elka0711, 2019-02-27:
That's the Trace-Log from UaExpert
[UaExpertTrace.log](https://github.com/FreeOpcUa/opcua-asyncio/files/2909012/UaExpertTrace.log)

cbergmiller, 2019-04-18:
@elka0711 Any Progress on this issue? Did you hear back from the vendor/maker?

cbergmiller, 2019-04-20:
I am assuming this problem was resolved or is caused by the server you were using. Feel free to open a new issue otherwise.

closed 2019-04-20 by cbergmiller