FreeOpcUa/opcua-asyncio #25: scrutinizer report

oroulet, 2019-03-15:
@cbergmiller looks like scrutinizer code checker does not like you.

[image]

 https://scrutinizer-ci.com/g/FreeOpcUa/opcua-asyncio/issues/master?orderField=issueCount&order=desc&honorSelectedPaths=0&page=1

cbergmiller, 2019-03-15:
That's unfortunate!
Many issues stem from my package refactoring.
e.g.:
`__all__ = (callback.__all__ + connection.__all__ + copy_node_util.__all__ + event_objects.__all__)`
It's valid syntax and a style recommended by David Beazley:
[https://www.dabeaz.com/modulepackage/ModulePackage.pdf](https://www.dabeaz.com/modulepackage/ModulePackage.pdf)

Should we/I refactor this?

oroulet, 2019-03-15:
That pdf is too long for my brain. What does it say?

is the `__all__ `stuff the recommended way to work?

scrutinizer is not against `__all__`, it just complains that the setting of `__all__` uses variables imported through `from xx import *` so it cannot check. Maybe the goal will be to remove some of the `from xx import *` . A few are really necessary to merge modules, but many are not.

swamper123, 2020-12-16:
Seems to be solved with the PR. Case closed 🕵️

closed 2020-12-16 by swamper123