FreeOpcUa/opcua-asyncio #26: BadWriteNotSupported

jylilaht, 2019-03-18:
I am trying to write value to opcua.

Basically my code is like this in example:

var = client.get_node("ns=3;i=2002")
print(var)
var.get_data_value() # get value of node as a DataValue object
var.get_value() # get value of node as a python builtin
var.set_value(3.9) # set node value using implicit data type

but I get this error:
asyncua.ua.uaerrors._auto.BadWriteNotSupported: The server does not support writing the combination of value, status and
timestamps provided.(BadWriteNotSupported)

Should I be using this instead:
var.set_value(ua.Variant([23], ua.VariantType.Int64)) #set node value using explicit data type

If so, what is best way to make sure that VariantType is right? I can have bool, float, int16,int32 and string. Surely I can make my own function to check it I think, but is there better way?

oroulet, 2019-03-18:
Yes either specific type or even timestamp if server wants it

jylilaht, 2019-03-19:
This works

```
myvarnode = client.get_node(nodestring)
value = await myvarnode.get_value()
myvarianttype = await myvarnode.get_data_type_as_variant_type()
mydatatype = await myvarnode.get_data_type()
print("Old value: " + str(value))
print("Variant type : " + str(myvarianttype))
print("Data type : " + str(mydatatype))

print("Input new value: ")
newvalue = input(">> ")
 
dv = ua.DataValue(ua.Variant(float(newvalue), myvarianttype))

sleep(1)
print("Datavalue (write): " + str(dv))
await myvarnode.set_attribute(ua.AttributeIds.Value, dv)
```

I only wonder if I could get right type for input easily like myvarianttype. So I would not need this float(xxx) or int(xxx) etc.

closed 2019-03-19 by jylilaht

David-2D3FE9, 2021-11-10:
Thanks so much for this share!  

I was uncessfull, even with explicit method like:  
`var.set_value(ua.Variant(value=int(8), varianttype=ua.VariantType.Int16))`  
Event if i doublechecked type with `var.get_data_type_as_variant_type()`

Using` .set_attribute()` with a previous constuction of the DataValue this work perfectly 👍