FreeOpcUa/opcua-asyncio #28: dependency to context manager

oroulet, 2019-04-04:
@cbergmiller Why do we have dependency to https://pypi.org/project/asyncio-contextmanager/ ?
It seems to be used at one place in https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/server/history_sql.py
What was the issue that made it needed?

cbergmiller, 2019-04-04:
I used it to provide a context handler for opening/closing the DB connection. This could be refactored if you want to remove the dependency.

oroulet, 2019-04-04:
I do not really like to have a dependency for something which is more like sample code ;-)
and why do we need the context handler? there is only one thread and one task so we can just share one global connection, can't we?

cbergmiller, 2019-04-04:
I don't have much experience with low level DB access. The cursor could be opened on `init` and closed on `stop`.

PR #35 (2019-04-20): removed async_contextmanager dependency, use sqlite3 shortcut methods
Fixes #28 
I removed the contextmanager dependency. Using the sqlite3 shortcut methods described here [https://docs.python.org/3/library/sqlite3.html#using-sqlite3-efficiently](https://docs.python.org/3/library/sqlite3.html#using-sqlite3-efficiently) i removed explicit creation of the cursor altogether.
closed 2019-04-20 by oroulet