FreeOpcUa/opcua-asyncio #29: Kepware tag browsing issue

kbotha, 2019-04-15:
Hi

I'm running an OPC Server in python (server-example.py) and trying to connect to it using an OPC UA Client in Kepware (KEPServerEX6). The connection works but I get an error message when I browse for tags and try to add them. If I manually try to add them using their address (eg. ns=2;i=13) then it works fine.

Browsing for Python OPC Server works:
[image]

Connection settings (password is blank):
[image]

I can then browse for tags and it shows all of the tags that should be there:
[image]

If I then select one/many of the tags I get the following error message:
[image]

Afterwards I can manually go and and tags which works fine.

Am I missing something on the python side? Or is it maybe some Kepware related issue?

Some log from pycharm:
INFO:asyncua.server.internal_server:Created internal session ('127.0.0.1', 65001)
INFO:asyncua.server.internal_server:Create session request
INFO:asyncua.server.internal_server:get endpoint
INFO:asyncua.server.uaprocessor:sending create session response
INFO:asyncua.server.uaprocessor:Activate session request
INFO:asyncua.server.internal_server:activate session
INFO:asyncua.server.internal_server:Activated internal session ('127.0.0.1', 65001) for user User.Anonymous
INFO:asyncua.server.uaprocessor:sending read response
INFO:asyncua.server.uaprocessor:Read request
INFO:asyncua.server.uaprocessor:sending read response
INFO:asyncua.server.uaprocessor:Browse request
INFO:asyncua.server.uaprocessor:sending browse response
INFO:asyncua.server.uaprocessor:Browse request
INFO:asyncua.server.uaprocessor:sending browse response
INFO:asyncua.server.uaprocessor:Read request
INFO:asyncua.server.uaprocessor:sending read response
INFO:asyncua.server.uaprocessor:Read request
INFO:asyncua.server.uaprocessor:sending read response
INFO:asyncua.server.uaprocessor:Read request
INFO:asyncua.server.uaprocessor:sending read response
INFO:asyncua.server.uaprocessor:Read request
INFO:asyncua.server.uaprocessor:sending read response
INFO:asyncua.server.uaprocessor:Browse request
INFO:asyncua.server.uaprocessor:sending browse response
INFO:asyncua.server.uaprocessor:Read request
INFO:asyncua.server.uaprocessor:sending read response
INFO:asyncua.server.uaprocessor:Close session request
!! This comes after I try to add the tags from the browse menu
INFO:asyncua.server.internal_server:close session %s
INFO:asyncua.server.subscription_service:delete subscriptions: []
INFO:asyncua.server.uaprocessor:sending close session response
INFO:asyncua.server.binary_server_asyncio:processor returned False, we close connection from ('127.0.0.1', 65001)
INFO:asyncua.server.binary_server_asyncio:Lost connection from ('127.0.0.1', 65001), None
INFO:asyncua.server.uaprocessor:Cleanup client connection: ('127.0.0.1', 65001)
INFO:asyncua.server.internal_server:close session %s
INFO:asyncua.server.subscription_service:delete subscriptions: []

kbotha, 2019-04-16:
Kepware actually can't read and of the tags from the Python OPC Server

closed 2019-04-16 by kbotha

oroulet, 2019-04-16:
If it does not work then we should have an open bug report

kbotha, 2019-04-16:
Ok, I'll make one