FreeOpcUa/opcua-asyncio #30: Error when trying to set_value() of a tag

kbotha, 2019-04-15:
Hi

I'm getting an error when I try to set a value using set_value() in a client.

I have a Kepware UA OPC Server running and connecting using a simple UA Client in python

When I try to use:
`tag1.set_value(5)`
I get:
`asyncua.ua.uaerrors._auto.BadWriteNotSupported: The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)`

I read some of the previous issues, so I then tried a few more options:
`dv = ua.DataValue(5, ua.VariantType.UInt16)`
`dv.ServerTimestamp = datetime.now()`
`await tag1.set_value(dv)`
and
`await tag1.set_attribute(ua.AttributeIds.Value, ua.Variant(value=5))`

and a few other combinations but I keep getting the error:

`AttributeError: 'Variant' object has no attribute 'ua_types'`
or
`AttributeError: 'VariantType' object has no attribute 'ua_types'`

(I am able to set the value of the same tag using another UA Client)

Stacktrace
ERROR:asyncua:error
Traceback (most recent call last):
  File "K:/repos/opcua-asyncio/examples/client-example.py", line 72, in run
    await tag1.set_value(dv)
  File "C:\Miniconda2\envs\python_opc\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "K:\repos\opcua-asyncio\asyncua\common\node.py", line 228, in set_value
    await self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "C:\Miniconda2\envs\python_opc\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "K:\repos\opcua-asyncio\asyncua\common\node.py", line 273, in set_attribute
    result = await self.server.write(params)
  File "C:\Miniconda2\envs\python_opc\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "K:\repos\opcua-asyncio\asyncua\client\ua_client.py", line 340, in write
    data = await self.protocol.send_request(request)
  File "C:\Miniconda2\envs\python_opc\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "K:\repos\opcua-asyncio\asyncua\client\ua_client.py", line 120, in send_request
    future = self._send_request(request, callback, timeout, message_type)
  File "K:\repos\opcua-asyncio\asyncua\client\ua_client.py", line 99, in _send_request
    binreq = struct_to_binary(request)
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 247, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 270, in to_binary
    return struct_to_binary(val)
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 242, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 282, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 282, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 270, in to_binary
    return struct_to_binary(val)
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 247, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 260, in to_binary
    return pack_uatype(vtype, val)
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 179, in pack_uatype
    return struct_to_binary(value)
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 247, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 260, in to_binary
    return pack_uatype(vtype, val)
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 179, in pack_uatype
    return struct_to_binary(value)
  File "K:\repos\opcua-asyncio\asyncua\ua\ua_binary.py", line 239, in struct_to_binary
    for name, uatype in obj.ua_types:
AttributeError: 'VariantType' object has no attribute 'ua_types'
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection

johns10, 2019-04-15:
This is probably associated with your specific opc server.  Kepware
specifically has this issue. I had to comment out the code where it was
setting the server timestamp to make it work with kepware

oroulet, 2019-04-15:
Try to set ServerTimestsamp to None or ClientTimestamp

kbotha, 2019-04-15:
@oroulet When I set ServerTimestsamp to None I get the error:
`AttributeError: 'VariantType' object has no attribute 'ua_types'`
(stack trace above)

oroulet, 2019-04-15:
I jsut tested and it works fine:

                dv = ua.DataValue(ua.Variant(1.2, ua.VariantType.Double))
                dv.ServerTimestamp = None
                dv.SourceTimestamp = None
                await var.set_value(dv)

kbotha, 2019-04-16:
It works :-)

I was doing `dv = ua.DataValue(5, ua.VariantType.UInt16)`
Instead of `dv = ua.DataValue(ua.Variant(5, ua.VariantType.UInt16))`
(which doesn't make sense now that I look at DataValue class `def __init__(self, variant=None, status=None)`)

Thanks for the help!

closed 2019-04-16 by kbotha

jonatanallan2k11, 2021-11-01:
Has anyone been able to write Int32 and Double values in Codesys?
didn't work:
 dv = ua.DataValue(ua.Variant(1.2, ua.VariantType.Double))
            dv.ServerTimestamp = None
            dv.SourceTimestamp = None
            await var.set_value(dv)

And
 dv = ua.DataValue(ua.Variant(1.2, ua.VariantType.Double))
            dv.ServerTimestamp = datetime.utcnow()
            dv.SourceTimestamp = datetime.utcnow()
            await var.set_value(dv)
also didn't work
Tks :)

AndreasHeine, 2021-11-01:
what error did you get?

westbullman, 2022-09-05:
dv = ua.DataValue(ua.Variant(1, ua.VariantType.Word))
thisResultItem.set_value(dv)

this is not OK，can anyone help me, thanks。

AndreasHeine, 2022-09-05:
@westbullman wouldn't it be better to make a own issue then posting on a 3 Year old one which is closed!?