FreeOpcUa/opcua-asyncio #32: No access to server with Bosch Rexroth OPC UA test client

M-Reimer, 2019-04-17:
I've started one of your simple server examples.
I have easy access with your own provided client.
I also have access with some Android Apps.
If I try with the test client, provided by Bosch Rexroth, then I can't manage to get a connection to my server.
https://www.boschrexroth.com/en/xc/products/product-groups/electric-drives-and-controls/industrial-iot/iot-gateway
You have to scroll down to "Downloads" and download the OPC UA Client from there.
I was not able to find out why the connection always fails but I think you have better ways of debugging.

oroulet, 2019-04-24:
you need to send code and  stack track of your test client. and most probably, if the server is not happy, record two sessions with wireshark. one with opcua-asyncio and one with a client that works (preferably uaexpert) then we will try to compare the differences

M-Reimer, 2019-04-24:
The client is available free of charge. I used it in a project for my employer but replaced it by your client.

I thought it may be interesting for you to test your server against some example client which is used "in the field".

I won't be able to help you as I privately don't have any Windows machines to run the client on.

oroulet, 2019-04-24:
OK. I do not have any windows PC either. so we'll wait until someone has the possibility to record a session

cbergmiller, 2019-04-28:
I tried to run the Bosch Rexroth client but it is an unsigned application that Windows Defender advises you not to run. I will not investigate this any further.
Thanks for reporting this. I will close this issue for now. Feel free to open a new issue if you encounter a problem when developing with production servers/clients.

closed 2019-04-28 by cbergmiller

oroulet, 2019-04-28:
@cbergmiller I thin we should keep that kind of bugs open. This is a clear an reproducible bug, most probably not a user error. And it is very important to support as much as possible of these strange devices. Although it might well be an error on Bosch side. The last interoperability bug I saw was a clear defect on Siemens side (SIMATIC product)

M-Reimer, 2019-04-28:
Some of the DLL files that come with the Bosch Rexroth client had some signs that the whole software is based around some externally developed library.

I just thought it may be interesting for you to know that some software, that is used by an company to test OPC-UA servers, doesn't work right with your server.

oroulet, 2019-04-28:
@M-Reimer but if you had the possibility to record a session and write down the output of the python server here, it would be great. I have currently absolluetely not windows PC available, neither at home nor at work

M-Reimer, 2019-04-28:
Me neither. I don't know if this software would run with Wine on a Linux machine.

I'm also a bit busy recently so I won't be able to set up all the stuff needed to record a session. Especially with zero knowledge about Wireshark.