FreeOpcUa/opcua-asyncio #33: UaStatusCodeError(self.value) after a fixed time

jorisvankessel, 2019-04-17:
Hi guys,

I am trying to make the opcua subscription working with the code below. The code is just the same as is in the example folder. (ofcourse there are some adjustments for making it work on my hardware)  But after a fixed time, approximately 10 secs it comes with the following error.

When I play with the sleep (`await asyncio.sleep(10)`) time, for example putting it to 500, it keeps on running for about 8 minutes before it will show up with the same error. 

Is this something on my site wrong or can it be a issue/bug?

```
import time
import asyncio
import logging

from asyncua import Client

logging.basicConfig(level=logging.INFO)
_logger = logging.getLogger('asyncua')
import datetime

class SubHandler(object):
    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("New data change event", node, val, data)

    def event_notification(self, event):
        print("New event", event)

async def run():
    url = "opc.tcp://192.168.150.1:4840/"
    try:
        async with Client(url=url) as client:
            root = client.get_root_node()
            _logger.info("Root node is: %r", root)
            objects = client.get_objects_node()
            _logger.info("Objects node is: %r", objects)

            # Node objects have methods to read and write node attributes as well as browse or populate address space
            _logger.info("Children of root are: %r", await root.get_children())

            uri = "urn:OMRON:NxOpcUaServer:FactoryAutomation"
            idx = await client.get_namespace_index(uri)
            _logger.info("index of our namespace is %s", idx)

            Counter = await root.get_child(["0:Objects", f"{idx}:new_Controller_0", "3:GlobalVars", "4:Counter"])

            obj = await root.get_child(["0:Objects", "4:new_Controller_0", "3:GlobalVars",])
            _logger.info("myvar is: %r", Counter)

            handler = SubHandler()
            sub = await client.create_subscription(500, handler)
            vars = [Counter]

            handle = await sub.subscribe_data_change(vars)
            await asyncio.sleep(10)

    except Exception:
        _logger.exception('error')

tijd_gestart = datetime.datetime.now()
_logger.info("Time start: {}".format(tijd_gestart))

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.set_debug(False)
    loop.run_until_complete(run())
    tijd_gestopt = datetime.datetime.now()
    _logger.info("Tijdsduur: {}".format(tijd_gestopt-tijd_gestart))
```

> uatypes.py

```
    def check(self):
        """
        Raises an exception if the status code is anything else than 0 (good).

        Use the is_good() method if you do not want an exception.
        """
        if not self.is_good():
            print("Value",self.value)
            raise UaStatusCodeError(self.value)
```
	
Python Console
==============
```
Value 2149974016
INFO:asyncua.client.ua_client.UaClient:call_publish_callback
WARNING:asyncua.client.ua_client.UASocketProtocol:ServiceFault from server received while waiting for publish response
ERROR:asyncio:Task exception was never retrieved
future: <Task finished coro=<UaClient._call_publish_callback() done, defined at D:\projects\Havatec\Intel RealSense\opc\lib\site-packages\asyncua\client\ua_client.py:468> exception=BadSessionClosed(2149974016)>
Traceback (most recent call last):
  File "D:\projects\Havatec\Intel RealSense\opc\lib\site-packages\asyncua\client\ua_client.py", line 471, in _call_publish_callback
    self.protocol.check_answer(data, "while waiting for publish response")
  File "D:\projects\Havatec\Intel RealSense\opc\lib\site-packages\asyncua\client\ua_client.py", line 133, in check_answer
    hdr.ServiceResult.check()
  File "D:\projects\Havatec\Intel RealSense\opc\lib\site-packages\asyncua\ua\uatypes.py", line 225, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
INFO:asyncua:Tijdsduur: 0:00:10.078220	
```

cbergmiller, 2019-04-18:
Hello @jorisvankessel, thanks for reporting this issue.
When the `asyncio.sleep(10)` finished the `Client` will disconnect because the code exits the context manager. So my first guess is that a problem occurs while disconnecting from the Server. I tried but could not replicate the issue with a Siemens S7 OPC UA Server. What server did you use to run the example?

jorisvankessel, 2019-04-18:
Hi @cbergmiller, thanks for your reply.
At this moment I use an OMRON NX PLC with a OPC UA Server.

cbergmiller, 2019-04-18:
Can you record the network traffic with Wireshark while the example ist running and attach the log?

jorisvankessel, 2019-04-18:
Here you go:
[opc_wireshark.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/3094398/opc_wireshark.zip)

cbergmiller, 2019-04-19:
The Server answers with a `ServiceFault`:
[image]
This should be handled when closing the session:
```python
    async def close_session(self, delete_subscriptions):
        self.logger.info("close_session")
        if self.protocol and self.protocol.state == UASocketProtocol.CLOSED:
            self.logger.warning("close_session was called but connection is closed")
            return
        request = ua.CloseSessionRequest()
        request.DeleteSubscriptions = delete_subscriptions
        data = await self.protocol.send_request(request)
        response = struct_from_binary(ua.CloseSessionResponse, data)
        try:
            response.ResponseHeader.ServiceResult.check()
        except BadSessionClosed:
            # Problem: closing the session with open publish requests leads to BadSessionClosed responses
            #          we can just ignore it therefore.
            #          Alternatively we could make sure that there are no publish requests in flight when
            #          closing the session.
            pass
```
However it looks like your Server answers the last `PublishRequest` with a `ServiceFault` - this case is not handled. We will have to check the OPC UA spec.
For now you could ignore the `BadSessionClosed` exception as a workaround. As i mentioned before the session will be closed when you exit the `Client` context manager, this is intentional.

oroulet, 2019-04-19:
Yes some servers send residual publish requests after close with service fault. I

jorisvankessel, 2019-04-19:
Great, I will implement the workaround for now. Thanks!

cbergmiller, 2019-04-19:
@oroulet this could be solved by setting a flag that indicates teardown of the session is in progress before sending the `CloseSessionRequest`. `ServiceFault` could then be ignored (when received as a response to a `PublishRequest`) when the flag is set.

oroulet, 2019-04-19:
I kind of remember, it was not so easy to fix. But I have not looked at that code for many years

jorisvankessel, 2019-04-22:
Ok, I tried to ignore the `BadSessionClosed` as you explained above. The session will always be closed in this stadium? I'm not sure what else to do for keep the program running. 

Thanks for advising so far.

```
INFO:asyncua.common.subscription:Publish callback called with result: PublishResult(SubscriptionId:3137412008, AvailableSequenceNumbers:[10], MoreNotifications:False, NotificationMessage:NotificationMessage(SequenceNumber:10, PublishTime:2019-04-22 19:49:27.084079, NotificationData:[DataChangeNotification(MonitoredItems:[MonitoredItemNotification(ClientHandle:201, Value:DataValue(Value:Variant(val:65,type:VariantType.Int16), StatusCode:StatusCode(Good), SourceTimestamp:2019-04-22 19:49:26.771079, ServerTimestamp:2019-04-22 19:49:26.771079))], DiagnosticInfos:[])]), Results:[StatusCode(Good)], DiagnosticInfos:[])
INFO:asyncua.client.ua_client.UaClient:publish
INFO:asyncua.client.client:disconnect
INFO:asyncua.client.ua_client.UaClient:close_session
INFO:asyncua.client.ua_client.UaClient:call_publish_callback
WARNING:asyncua.client.ua_client.UASocketProtocol:ServiceFault from server received while waiting for publish response
ERROR:asyncio:Task exception was never retrieved
future: <Task finished coro=<UaClient._call_publish_callback() done, defined at D:\projects\opcua\lib\site-packages\asyncua\client\ua_client.py:470> exception=BadSessionClosed(2149974016)>
Traceback (most recent call last):
  File "D:\projects\opcua\lib\site-packages\asyncua\client\ua_client.py", line 473, in _call_publish_callback
    self.protocol.check_answer(data, "while waiting for publish response")
  File "D:\projects\opcua\lib\site-packages\asyncua\client\ua_client.py", line 133, in check_answer
    hdr.ServiceResult.check()
  File "D:\projects\opcua\lib\site-packages\asyncua\ua\uatypes.py", line 224, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection

```

cbergmiller, 2019-04-23:
@jorisvankessel what are you trying to do? Do you want to keep a client connection open? You can write your code so that is doesn't leave the context manager (e.g. while loop). Or you could use Client.connect() /.disconnect() directly without the context manager.

jorisvankessel, 2019-04-23:
@cbergmiller When the BadSessionClosed error appears I wil log this occurrence, but I would like to keep my program running and keep on logging so it won't interference any other process that's running.

cbergmiller, 2019-04-23:
I am a bit confused - I am pretty sure the `BadSessionClosed` Expception is raised because the asyncua client (your program) tries to close the connection. Do you think the OPC UA connection is closed unexpectedly? I am not sure what _I would like to keep my program running_ refers to without more knowledge about your applications architecture.

jorisvankessel, 2019-04-23:
Ok not a problem, I will try to explain a littlebit more about the purpose. In my program I subscribe to several tags available in the OPC UA server on the PLC. I store these tagvalues into a database for further processing. This _program_ needs to be running all the time consuming these values. But in my situation the connection will be disconnected after about 10secs because of this `BadSessionClosed `occurence. In this case I would like to _ignore_ the exception like you suggested and go on with sampling the tagvalues. Somehow ignoring the exception will stil disconnect my OPC UA connection.

cbergmiller, 2019-04-23:
If your write:
```python
async with Client(url=url) as client:
    # Do something with client
    await asyncio.sleep(10)
```
The connection will be closed after 10 seconds. This is intentional.
Either use something like this:
```python
async with Client(url=url) as client:
    while True:
        # Do something with client
```

Or open the client connection without context manager:
```python
client = asyncua.Client(url=url)
await client.connect()
while True:
    # do something with client
# close connection before you exit the program
await client.disconnect()
```

I hope this helps.

jorisvankessel, 2019-04-23:
@cbergmiller I was working on your first suggestion, hoping this was the right direction. I will test it, for now thank you for your quick response and thinking along with this.

1462326016, 2019-04-25:
I also encountered this problem, but when I executed `await sub.unsubscribe(handle);
await sub.delete() `after executing `await asyncio.sleep(10)`, it works fine if I don't cancel the subscription.The UA server I am using is kepserver.

cbergmiller, 2019-04-28:
@1462326016 you also get a `BadSessionClosed` exception when closing a `asyncua.Client` connection?
Is this a problem in your application? We need more context (code example, stack trace) to diagnose if this is a related issue.

1462326016, 2019-04-28:
`import asyncio

from asyncua import Client

# logging.basicConfig(level=logging.INFO)

class SubHandler(object):
    def datachange_notification(self, node, val, data):
        # print("New data change event", node, val, data)
        print("New data change event", node, val)

    def event_notification(self, event):
        print("New event", event)

async def run():
    url = "opc.tcp://127.0.0.1:4840"
    async with Client(url=url) as client:
        handler = SubHandler()
        sub = await client.create_subscription(500, handler)
        my = client.get_node('ns=2;s=test.test.alarm.alarm1')
        await sub.subscribe_data_change([my])
        await asyncio.sleep(5)
        sub.unsubscribe(handler)
        del sub

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    # loop.set_debug(True)
    loop.run_until_complete(run())
`
My code is here.
`C:\Python3\python.exe C:/work/async/client-example.py
New data change event Node(StringNodeId(ns=2;s=test.test.alarm.alarm1)) False
<CoroWrapper Subscription.unsubscribe() running at C:\Python3\lib\site-packages\asyncua\common\subscription.py:272, created at C:\Python3\lib\asyncio\coroutines.py:85> was never yielded from
Coroutine object created at (most recent call last, truncated to 10 last lines):
  File "C:/work/async/client-example.py", line 32, in <module>
    loop.run_until_complete(run())
  File "C:\Python3\lib\asyncio\base_events.py", line 471, in run_until_complete
    self.run_forever()
  File "C:\Python3\lib\asyncio\base_events.py", line 438, in run_forever
    self._run_once()
  File "C:\Python3\lib\asyncio\base_events.py", line 1443, in _run_once
    handle._run()
  File "C:\Python3\lib\asyncio\events.py", line 145, in _run
    self._callback(*self._args)
  File "C:/work/async/client-example.py", line 25, in run
    sub.unsubscribe(handler)
  File "C:\Python3\lib\asyncio\coroutines.py", line 85, in debug_wrapper
    return CoroWrapper(gen, None)
C:/work/async/client-example.py:25: RuntimeWarning: coroutine 'Subscription.unsubscribe' was never awaited
  sub.unsubscribe(handler)
ServiceFault from server received while waiting for publish response
Task exception was never retrieved
future: <Task finished coro=<UaClient._call_publish_callback() done, defined at C:\Python3\lib\site-packages\asyncua\client\ua_client.py:468> exception=BadSessionClosed(2149974016,) created at C:\Python3\lib\site-packages\asyncua\client\ua_client.py:447>
source_traceback: Object created at (most recent call last):
  File "C:/work/async/client-example.py", line 32, in <module>
    loop.run_until_complete(run())
  File "C:\Python3\lib\asyncio\base_events.py", line 471, in run_until_complete
    self.run_forever()
  File "C:\Python3\lib\asyncio\base_events.py", line 438, in run_forever
    self._run_once()
  File "C:\Python3\lib\asyncio\base_events.py", line 1443, in _run_once
    handle._run()
  File "C:\Python3\lib\asyncio\events.py", line 145, in _run
    self._callback(*self._args)
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 447, in _sub_data_received
    self.loop.create_task(self._call_publish_callback(data))
Traceback (most recent call last):
  File "C:\Python3\lib\asyncio\coroutines.py", line 126, in send
    return self.gen.send(value)
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 471, in _call_publish_callback
    self.protocol.check_answer(data, "while waiting for publish response")
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 133, in check_answer
    hdr.ServiceResult.check()
  File "C:\Python3\lib\site-packages\asyncua\ua\uatypes.py", line 224, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)

进程已结束，退出代码 0
`
Stack trace is here.

1462326016, 2019-04-28:
> @1462326016 you also get a `BadSessionClosed` exception when closing a `asyncua.Client` connection?
> Is this a problem in your application? We need more context (code example, stack trace) to diagnose if this is a related issue.

An exception occurs when I run the code for 5 seconds.Is my method correct? I don't know how to upload files. If you need an .opf file, I can send it to you.My version of kepserver is V6.4.321.0

cbergmiller, 2019-04-28:
@1462326016 in your example you didn't await sub.unsubscribe(handler).  This causes the Exception:
`RuntimeWarning: coroutine 'Subscription.unsubscribe' was never awaited
  sub.unsubscribe(handler)`
Can you change that and try again?

1462326016, 2019-04-28:
> @1462326016 in your example you didn't await sub.unsubscribe(handler). This causes the Exception:
> `RuntimeWarning: coroutine 'Subscription.unsubscribe' was never awaited sub.unsubscribe(handler)`
> Can you change that and try again?

Thank you very much for replying me so quickly.I modified my program.But he still throws an exception.Stack trace below.I am not sure if it is my problem, it seems to have an error in parsing the binary.
```import asyncio

from asyncua import Client

# logging.basicConfig(level=logging.INFO)

class SubHandler(object):
    def datachange_notification(self, node, val, data):
        # print("New data change event", node, val, data)
        print("New data change event", node, val)

    def event_notification(self, event):
        print("New event", event)

async def run():
    url = "opc.tcp://127.0.0.1:4840"
    async with Client(url=url) as client:
        handler = SubHandler()
        sub = await client.create_subscription(500, handler)
        my = client.get_node('ns=2;s=test.test.alarm.alarm1')
        await sub.subscribe_data_change([my])
        await asyncio.sleep(5)
        await sub.unsubscribe(handler)
        await sub.delete()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    # loop.set_debug(True)
    loop.run_until_complete(run())
```
```C:\Python3\python.exe C:/work/async/client-example.py
New data change event Node(StringNodeId(ns=2;s=test.test.alarm.alarm1)) False
ServiceFault from server received while waiting for publish response
Task exception was never retrieved
future: <Task finished coro=<UaClient._call_publish_callback() done, defined at C:\Python3\lib\site-packages\asyncua\client\ua_client.py:468> exception=BadSessionClosed(2149974016,) created at C:\Python3\lib\site-packages\asyncua\client\ua_client.py:447>
source_traceback: Object created at (most recent call last):
  File "C:/work/async/client-example.py", line 32, in <module>
    loop.run_until_complete(run())
  File "C:\Python3\lib\asyncio\base_events.py", line 471, in run_until_complete
    self.run_forever()
  File "C:\Python3\lib\asyncio\base_events.py", line 438, in run_forever
    self._run_once()
  File "C:\Python3\lib\asyncio\base_events.py", line 1443, in _run_once
    handle._run()
  File "C:\Python3\lib\asyncio\events.py", line 145, in _run
    self._callback(*self._args)
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 447, in _sub_data_received
    self.loop.create_task(self._call_publish_callback(data))
Traceback (most recent call last):
  File "C:\Python3\lib\asyncio\coroutines.py", line 126, in send
    return self.gen.send(value)
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 471, in _call_publish_callback
    self.protocol.check_answer(data, "while waiting for publish response")
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 133, in check_answer
    hdr.ServiceResult.check()
  File "C:\Python3\lib\site-packages\asyncua\ua\uatypes.py", line 224, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)
Traceback (most recent call last):
  File "C:/work/async/client-example.py", line 32, in <module>
    loop.run_until_complete(run())
  File "C:\Python3\lib\asyncio\base_events.py", line 484, in run_until_complete
    return future.result()
  File "C:/work/async/client-example.py", line 25, in run
    await sub.unsubscribe(handler)
  File "C:\Python3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Python3\lib\site-packages\asyncua\common\subscription.py", line 280, in unsubscribe
    results = await self.server.delete_monitored_items(params)
  File "C:\Python3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 531, in delete_monitored_items
    data = await self.protocol.send_request(request)
  File "C:\Python3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 120, in send_request
    future = self._send_request(request, callback, timeout, message_type)
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 99, in _send_request
    binreq = struct_to_binary(request)
  File "C:\Python3\lib\site-packages\asyncua\ua\ua_binary.py", line 246, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "C:\Python3\lib\site-packages\asyncua\ua\ua_binary.py", line 269, in to_binary
    return struct_to_binary(val)
  File "C:\Python3\lib\site-packages\asyncua\ua\ua_binary.py", line 241, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "C:\Python3\lib\site-packages\asyncua\ua\ua_binary.py", line 279, in list_to_binary
    return data_type.pack_array(val)
  File "C:\Python3\lib\site-packages\asyncua\ua\ua_binary.py", line 132, in pack_array
    return size_data + struct.pack(self._fmt.format(len(data)), *data)
struct.error: required argument is not an integer

进程已结束，退出代码 1
```

cbergmiller, 2019-04-28:
@1462326016 Subscription.unsubscribe() requires you to pass a handle that was returned by Subscription.subscribe_data_change().
Since Subscription.subscribe_data_change() returns a list of handles you would have to call unsubscribe multiple times.
I suggest you use `await subscription.delete()` if you want to remove all Subscriptions - it's easier to use.

1462326016, 2019-04-28:
> @1462326016 Subscription.unsubscribe() requires you to pass a handle that was returned by Subscription.subscribe_data_change().
> Since Subscription.subscribe_data_change() returns a list of handles [...]

I used `await subscription.delete()` and it works fine.
Thank you very much for your help. I have encountered a new problem when I get the node. Can you give me some help? [https://github.com/FreeOpcUa/opcua-asyncio/issues/40](url)

closed 2019-05-15 by oroulet