FreeOpcUa/opcua-asyncio #36: need cleaner module/class import in __init__.py(s) files

oroulet, 2019-04-20:
Scrutinizer complains a lot about the code in the __init__,py files. @cbergmiller you wrote/changed them. I am not sure how they should be. Someone should find out what the best practices are...
https://scrutinizer-ci.com/g/FreeOpcUa/opcua-asyncio/

oroulet, 2019-04-20:
My understanding is that it is better to have as little code as possible in these files. as the libray grows (And it is already quite big) importing these modules cost
Only the stuff used absolutely everywhere llike Node should be imported there
https://stackoverflow.com/questions/35727134/module-imports-and-init-py-in-python

cbergmiller, 2019-04-20:
I can change the import strategy and remove the `__all__` magic. Should improve the Scrutinizer score.

oroulet, 2019-04-20:
The goal is not to improve the score, I am just thinking that it probably highlight an issue ;-)

closed 2019-04-24 by oroulet