FreeOpcUa/opcua-asyncio #40: I got an error when I got a lot of nodes  asyncua.ua.uaerrors._base.UaError: Wrong sequence 518 -> 518 (server bug or replay attack)

1462326016, 2019-04-28:
I got an error when I got a lot of nodes.
This error is not necessarily happening,It may take a few more runs to happen.
I have a device and thousands of nodes in my kepserver.

```import asyncio
import re

from asyncua import Client

async def main():
    url = 'opc.tcp://localhost:4840'
    r = re.compile(r'^_.+?|.+?\._.+?')
    async with Client(url=url) as client:  # type:Client
        for a in await client.get_objects_node().get_children():
            if isinstance(a.nodeid.Identifier, int) or r.match(a.nodeid.Identifier):
                continue
            print(f'111{a}')
            for b in await a.get_children():
                if isinstance(b.nodeid.Identifier, int) or r.match(b.nodeid.Identifier):
                    continue
                print(f'222{b}')
                for c in await b.get_children():
                    if isinstance(c.nodeid.Identifier, int) or r.match(c.nodeid.Identifier):
                        continue
                    print(f'333{c}')
                    for d in await c.get_children():
                        if isinstance(d.nodeid.Identifier, int) or r.match(d.nodeid.Identifier):
                            continue
                        # print(f'444{d}')

if __name__ == '__main__':
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
    loop.close()
```
**stack trace**
```111Node(StringNodeId(ns=2;s=Collection))
222Node(StringNodeId(ns=2;s=Collection.ANDON1))
333Node(StringNodeId(ns=2;s=Collection.ANDON1.alarm))
Exception raised while parsing message from server
Traceback (most recent call last):
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 68, in _process_received_data
    msg = self._connection.receive_from_header_and_body(header, buf)
  File "C:\Python3\lib\site-packages\asyncua\common\connection.py", line 282, in receive_from_header_and_body
    return self._receive(chunk)
  File "C:\Python3\lib\site-packages\asyncua\common\connection.py", line 299, in _receive
    self._check_incoming_chunk(msg)
  File "C:\Python3\lib\site-packages\asyncua\common\connection.py", line 262, in _check_incoming_chunk
    .format(self._peer_sequence_number, num)
asyncua.ua.uaerrors._base.UaError: Wrong sequence 518 -> 518 (server bug or replay attack)
close_session was called but connection is closed
close_secure_channel was called but connection is closed
disconnect_socket was called but connection is closed
Traceback (most recent call last):
  File "C:/work/async/client-minimal.py", line 31, in <module>
    loop.run_until_complete(main())
  File "C:\Python3\lib\asyncio\base_events.py", line 484, in run_until_complete
    return future.result()
  File "C:/work/async/client-minimal.py", line 23, in main
    for d in await c.get_children():
  File "C:\Python3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Python3\lib\site-packages\asyncua\common\node.py", line 403, in get_referenced_nodes
    references = await self.get_references(refs, direction, nodeclassmask, includesubtypes)
  File "C:\Python3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Python3\lib\site-packages\asyncua\common\node.py", line 382, in get_references
    results = await self.server.browse(params)
  File "C:\Python3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 300, in browse
    data = await self.protocol.send_request(request)
  File "C:\Python3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 122, in send_request
    await asyncio.wait_for(future, timeout if timeout else None)
  File "C:\Python3\lib\asyncio\tasks.py", line 362, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError

```

1462326016, 2019-04-28:
I will upload a wireshark session later.

oroulet, 2019-04-28:
How many nodes? How does your code looks like? It math simply be that you overload the server. The timeout can also be increased. I think this is a parameter to Client

1462326016, 2019-04-29:
I have 797 nodes.

I think it is because this error caused a timeout.`asyncua.ua.uaerrors._base.UaError: Wrong sequence 518 -> 518 (server bug or replay attack)`

This error is not necessarily happening,It may take a few more runs to happen.

**stack trace**

```222Node(StringNodeId(ns=2;s=Collection.ANDON1))
333Node(StringNodeId(ns=2;s=Collection.ANDON1.alarm))
Exception raised while parsing message from server
Traceback (most recent call last):
  File "C:\Python3\lib\site-packages\asyncua\client\ua_client.py", line 68, in _process_received_data
    msg = self._connection.receive_from_header_and_body(header, buf)
  File "C:\Python3\lib\site-packages\asyncua\common\connection.py", line 282, in receive_from_header_and_body
    return self._receive(chunk)
  File "C:\Python3\lib\site-packages\asyncua\common\connection.py", line 299, in _receive
    self._check_incoming_chunk(msg)
  File "C:\Python3\lib\site-packages\asyncua\common\connection.py", line 262, in _check_incoming_chunk
    .format(self._peer_sequence_number, num)
asyncua.ua.uaerrors._base.UaError: Wrong sequence 518 -> 518 (server bug or replay attack)
```

**My code is**

```import asyncio
import re

from asyncua import Client

async def main():
    url = 'opc.tcp://localhost:4840'
    r = re.compile(r'^_.+?|.+?\._.+?')
    async with Client(url=url) as client:  # type:Client
        for a in await client.get_objects_node().get_children():
            if isinstance(a.nodeid.Identifier, int) or r.match(a.nodeid.Identifier):
                continue
            print(f'111{a}')
            for b in await a.get_children():
                if isinstance(b.nodeid.Identifier, int) or r.match(b.nodeid.Identifier):
                    continue
                print(f'222{b}')
                for c in await b.get_children():
                    if isinstance(c.nodeid.Identifier, int) or r.match(c.nodeid.Identifier):
                        continue
                    print(f'333{c}')
                    for d in await c.get_children():
                        if isinstance(d.nodeid.Identifier, int) or r.match(d.nodeid.Identifier):
                            continue
                        # print(f'444{d}')

if __name__ == '__main__':
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
    loop.close()
```

cbergmiller, 2019-04-29:
@1462326016 we will look into this. Until then some quick thoughts:
- 800 Nodes is not that much.  It is possible to traverse over thousands of nodes without any problem
- If you are writing code for production you probably want to make it fault tolerant (e.g. when your connection is closed unexpectedly try again after a short wait)

oroulet, 2019-04-29:
@cbergmiller I have see this one before: """Wrong sequence 518 -> 518 (server bug or replay attack)""" There is probably a bug or weakness in the packet reconstruction code. but I am not the author and never really looked at it

1462326016, 2019-04-29:
> @1462326016 we will look into this. Until then some quick thoughts:
>
> * 800 Nodes is not that much.  It is possible to traverse over thousands of nodes without any problem
> * If you are writing code [...]

I originally had thousands of nodes,I reduced it to 800 nodes in order to determine the problem.I can't go on because there will still be errors after reconnecting.

Ilyazyk, 2019-05-03:
I have the same error with Rockwell server on line 
`await client.get_namespace_index(uri)`
It happens not always. 
My code is:

```
import asyncio
import sys
sys.path.insert(0, "..")
import logging
from asyncua import Client, Node, ua

logging.basicConfig(level=logging.DEBUG)
_logger = logging.getLogger("asyncua")

async def main():
    url = "opc.tcp://localhost:4990/FactoryTalkLinxGateway1"
    try:
        async with Client(url=url) as client:
            root = client.get_root_node()
            print('Children of root are:', await root.get_children())
             uri = "http://www.rockwellautomation.com/OpcUa"
            idx = await client.get_namespace_index(uri)
    except Exception:
        _logger.exception("error")

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(main())
    loop.close()
```
I have an error:

```
INFO:asyncua.client.ua_client.UaClient:read
DEBUG:asyncua.client.ua_client.UASocketProtocol:Sending: ReadRequest(TypeId:FourByteNodeId(i=631), RequestHeader:RequestHeader(AuthenticationToken:ByteStringNodeId(b=b'\xad\x8e\xd4\xb4@io&0\xf7~f\x80\xee\xf4\xc9\xfd\x1fzG_\x04\x9e\xb4|&\x12\xf2\xffZh\xbb'), Timestamp:2019-05-03 20:56:07.714032, RequestHandle:5, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:5000, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:ReadParameters(MaxAge:0, TimestampsToReturn:0, NodesToRead:[ReadValueId(NodeId:NumericNodeId(i=2255), AttributeId:13, IndexRange:None, DataEncoding:QualifiedName(0:None))]))
DEBUG:asyncio:poll 5000.000 ms took 0.000 ms: 1 events
DEBUG:asyncua.client.ua_client.UASocketProtocol:We did not receive enough data from server. Need 47781 got 43892
DEBUG:asyncio:poll 5000.000 ms took 0.000 ms: 1 events
ERROR:asyncua.client.ua_client.UASocketProtocol:Exception raised while parsing message from server
Traceback (most recent call last):
  File "..\asyncua\client\ua_client.py", line 78, in _process_received_data
    msg = self._connection.receive_from_header_and_body(header, buf)
  File "..\asyncua\common\connection.py", line 282, in receive_from_header_and_body
    return self._receive(chunk)
  File "..\asyncua\common\connection.py", line 299, in _receive
    self._check_incoming_chunk(msg)
  File "..\asyncua\common\connection.py", line 262, in _check_incoming_chunk
    .format(self._peer_sequence_number, seq_num)
asyncua.ua.uaerrors._base.UaError: Wrong sequence 5 -> 5 (server bug or replay attack)
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
```
An interesting thing is: if I delete the line "
`print('Children of root are:', await root.get_children())`,
I have "Wrong sequence 4 -> 4".

PR #48 (2019-05-04): Fix UASocketProtocol receive buffer
Fixes #40
Fragmented data reception could lead to `UASocketProtocol` processing the same message twice which must not happen and leads to a `UaError` because of the repeated sequence number.
closed 2019-05-04 by oroulet