FreeOpcUa/opcua-asyncio #42: lost exception in AttributeService.write

oroulet, 2019-04-29:
@cbergmiller I found this one: https://github.com/FreeOpcUa/opcua-asyncio/pull/41/commits/3126dc89d5aef8a1bc62fef83bf8bea8f0fad24e
This fixes one bug, BUT there is one more. If an exception is raised at that line in code, then the write call hangs forever, nothing printed in stdout and nothing is sent back to client. It looks like the exception is lost in some asyncio tasks?!?!?
If you have two minutes it would be great if you could have a look.

oroulet, 2019-04-29:
before that commit, you can provoke the bug by calling:
tools/uawrite -n i=85 -v DEBUG -t guess 4 (need fixes in https://github.com/FreeOpcUa/opcua-asyncio/pull/41/commits/461cb1aaf54b87965c83ff0931fe341defab8427 )

cbergmiller, 2019-04-29:
Is the exception only lost when using the tools?

oroulet, 2019-04-29:
Probably not. It probably happens for any error in address space

oroulet, 2019-04-29:
Any error generated by a client call

cbergmiller, 2019-04-29:
I found the problem(s):
1. Missing message param for logger.exception. Fixed in c5398c108a659cd2797f84c90bc029644279bb63
2. `uaprocessor` only sends service fault to client in case of `ServiceError` /commit/c5398c108a659cd2797f84c90bc029644279bb63
https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/server/uaprocessor.py#L123
The second problem i did not fix yet. Unsure what to do in order to respect the OPC UA Spec.

oroulet, 2019-04-29:
use logger.exception (not logger.info) at that place. Having an exception here is not normal.

cbergmiller, 2019-04-29:
But what to to when `_process_message` fails with an exception other than `ServiceError`. This should never happen but when it does there will be no response sent to the client.
Should we send a generic `ServiceFault`? I will have to look into the Spec.

oroulet, 2019-04-29:
Why do we only catch ServiceError here? should'nt we catch everything and send ServiceFault to client? ( I always understood ServiceFault as a kind of internal server error but maybe I am wrong)

oroulet, 2019-04-29:
but ideally we should answer to exactly that request, so clients can match the request to the error and raise error at the correct place

cbergmiller, 2019-04-29:
I added handling of general errors in uaprocessor. Now a `BadInternalError` is sent to the Client. Exception is logged with `logger.exception()`
Added in b770da4ce1aca030513475478c70d1d80cebd65e

closed 2019-05-15 by oroulet