FreeOpcUa/opcua-asyncio #56: Long-running loop sees internal subscription processing error

dannmartens, 2019-06-04:
Hi,

We're using a long-running client subscription loop which connects to a SIMATIC WinCC OA server in order to log events.

After several days, an error internal to the library is raised, which does not propagate to the subscriber handler. As a result, the entire process stops logging events, while the subscriber continues on its loop without a clue that something has gone wrong. 

We're using asyncua == 0.5.0.

The entire context is guarded to facilitate a clean exit in case of trouble:
```
        async with self.client:
            try:
            except:
```

Yet nothing gets through to break off the client subscription.

The only output is a short trace:
```
11:37:16 [WARNING] asyncua.uaprotocol - Received an error: MessageAbort(error:StatusCode(Good), reason:None)
11:37:16 [CRITICAL] asyncua.client.ua_client.UASocketProtocol - Received an error: MessageAbort(error:StatusCode(Good), reason:None)
11:37:16 [ERROR] asyncua.client.ua_client.UASocketProtocol - Exception raised while parsing message from server
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 69, in _process_received_data
    self._process_received_message(msg)
  File "/usr/local/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 86, in _process_received_message
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/usr/local/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 141, in _call_callback
    request_id, self._callbackmap.keys()))
asyncua.ua.uaerrors._base.UaError: No request found for requestid: 0, callbacks in list are dict_keys([7333, 7334])
```

We're a bit stumped. One option on the table is to fall back to the synchronous library, as this seems very black box. The last version published seems 0.5.0 on PyPI; has there been a more recent update we can try?

Your feedback would be appreciated!

Kind regards,
 Dann

oroulet, 2019-06-04:
interesting. we receive a MessageAbort from server. Then all the pending requests should go  quickly in timeout and raise an exception. Did you pass a very long timeout to client?

@cbergmiller looking at this code, a nice improvment could be to go though the entire _callbackmap and cancel all of them , thus all pending operations will raised a CancelledError

@dannmartens can you try master? I think there was a bug with connection renewal that has been fixed. that may explain the issue

@cbergmiller also did you notice that _callbackmap is very short... that version does not have the cleanup of that dict, so I was expecting it to be very long... what is happening here....

dannmartens, 2019-06-04:
Hi @oroulet,

I've redeployed after changing the dependency from:
    `asyncua == 0.5.0`
 to
    `git+https://github.com/FreeOpcUa/opcua-asyncio#egg=asyncua`

Because the version is burned into the source, it still claims to install 0.5.0, but I am pretty sure it only gets the head of current master from GitHub. We build wheels for everything first, including from the asyncua GitHub sourc,e and then only pip install the wheels we actually have.

It's up in staging right now, so let's wait and see.

Logging settings are sparse, we might need to open them up more if something keeps going wrong.

cbergmiller, 2019-06-05:
I think there may be still some flaws in `UASocketProtocol`. In case of an _Acknowledge_ or an _ErrorMessage_ `_call_callback` gets always called with an message id of zero. There is no special handling of message id zero in `_call_callback` though. There will be not future to resolve or cancel with id zero in the callback map.
The right thing to do may be to cancel all request futures (at least in case of a received _MessageAbort_).
I will open an issue for this this (the correct behaviour from the OPC UA spec. has to be implemented).

Another problem is the propagation of errors. If a non recoverable error occurs in the Publish task the main code will never see an Exception because callbacks are used to send Subscription Results to the main program. Maybe some kind of error message could be sent through the callbacks?

@oroulet The Request ID list is so short because the memory leak was introduced by me in d35d5e7421638b994cb4a6bab9dd363808e7165e The published version looks like this (notice the use of pop):
```python
def _call_callback(self, request_id, body):
        future = self._callbackmap.pop(request_id, None)
        if future is None:
            raise ua.UaError("No request found for requestid: {0}, callbacks in list are {1}".format(
                request_id, self._callbackmap.keys()))
    future.set_result(body)
```

oroulet, 2019-06-05:
>> Another problem is the propagation of errors. If a non recoverable error occurs in the Publish task the main code will never see an Exception because callbacks are used to send Subscription Results to the main program. Maybe some kind of error message could be sent through the callbacks?

I have not looked closely at that issue for a while, but isn'it good enough to just close the connection and cancel all tasks? if it is unrecoverable, we cannot do anything else anyway, if connection get closed it will be discovered at next action from main thread. A client that does not support missing some events must anyway do some polling in main thread to make sure everyhting i OK.

dannmartens, 2019-06-05:
Some early feedback: the process did not malfunction over a period of about 16 hours.

With the current HEAD of master integrated, we do get the following warnings:
```
2019-06-04 19:33:18 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 669, expected 668
2019-06-04 20:16:01 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 852, expected 851
2019-06-04 20:30:16 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 913, expected 912
2019-06-04 22:43:07 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 1482, expected 1481
2019-06-05 00:37:46 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 1973, expected 1972
2019-06-05 04:06:02 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 2865, expected 2864
2019-06-05 04:20:17 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 2926, expected 2925
```

oroulet, 2019-06-05:
the chunck stuff is another thing. the error message appear sometimes, but I have not seens error relaled to it (although in theory it should result in big bad parsing error)

cbergmiller, 2019-06-06:
The warnings are related to the `SecurityToken` that the server uses to encrypt/authenticate the message it sends to the client. The warning tells you that the client could not verify the token.
@dannmartens Can you tell us how often this warning appears (roughly)?

dannmartens, 2019-06-06:
The service has been running in staging for about 45 hours now (16 occurrences):
```
2019-06-04 19:33:18 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 669, expected 668
2019-06-04 20:16:01 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 852, expected 851
2019-06-04 20:30:16 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 913, expected 912
2019-06-04 22:43:07 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 1482, expected 1481
2019-06-05 00:37:46 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 1973, expected 1972
2019-06-05 04:06:02 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 2865, expected 2864
2019-06-05 04:20:17 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 2926, expected 2925
2019-06-05 10:45:32 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 4576, expected 4575
2019-06-05 11:41:48 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 4817, expected 4816
2019-06-05 11:56:03 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 4878, expected 4877
2019-06-05 12:09:35 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 4936, expected 4935
2019-06-05 15:19:25 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 5749, expected 5748
2019-06-05 15:47:40 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 5870, expected 5869
2019-06-05 15:52:20 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 5890, expected 5889
2019-06-05 20:01:55 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 6959, expected 6958
2019-06-06 02:00:31 [WARNING] asyncua.uaprotocol - Received a chunk with wrong token id 8495, expected 8494
```

What is odd is that the message is consistent with "wrong token id N+1, expected N." 

And here's another fun fact:
[image]

cbergmiller, 2019-06-06:
This is probably a timing issue that occurs when the security token of the session in renewed. The old token is wrongfully considered invalid by the client. I opened a separate issue https://github.com/FreeOpcUa/opcua-asyncio/issues/59 and will try to resolve it.
Thanks for reporting this problem!

kunz07, 2020-07-30:
I'm experiencing the same issue wherein a long standing connection breaks off with a BadSecureChannelTokenUnknown. The exception is not getting propagated even though it is encapsulated within a try catch block

```2020-07-30 06:32:57,431 [WARNING] Received an error: MessageAbort(error:StatusCode(BadSecureChannelTokenUnknown), reason:None)
2020-07-30 06:32:57,433 [CRITICAL] Received an error: MessageAbort(error:StatusCode(BadSecureChannelTokenUnknown), reason:None)
2020-07-30 06:32:57,434 [ERROR] Exception raised while parsing message from server
2020-07-30 06:32:57,433 [CRITICAL] Received an error: MessageAbort(error:StatusCode(BadSecureChannelTokenUnknown), reason:None)
2020-07-30 06:32:57,434 [ERROR] Exception raised while parsing message from server
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/asyncua/client/ua_client.py", line 152, in _call_callback
    self._callbackmap[request_id].set_result(body)
KeyError: 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/asyncua/client/ua_client.py", line 79, in _process_received_data
    self._process_received_message(msg)
  File "/usr/local/lib/python3.7/dist-packages/asyncua/client/ua_client.py", line 98, in _process_received_message
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/usr/local/lib/python3.7/dist-packages/asyncua/client/ua_client.py", line 155, in _call_callback
    f"No request found for request id: {request_id}, pending are {self._callbackmap.keys()}"
asyncua.ua.uaerrors._base.UaError: No request found for request id: 0, pending are dict_keys([14123, 14124])```

Any help would be much appreciated

Regards,
Kunal

kunz07, 2020-07-30:
I'm experiencing the same issue wherein a long standing connection breaks off with a BadSecureChannelTokenUnknown. The exception is not getting propagated even though it is encapsulated within a try catch block

```
2020-07-30 06:32:57,433 [CRITICAL] Received an error: MessageAbort(error:StatusCode(BadSecureChannelTokenUnknown), reason:None)
2020-07-30 06:32:57,434 [ERROR] Exception raised while parsing message from server
2020-07-30 06:32:57,433 [CRITICAL] Received an error: MessageAbort(error:StatusCode(BadSecureChannelTokenUnknown), reason:None)
2020-07-30 06:32:57,434 [ERROR] Exception raised while parsing message from server
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/asyncua/client/ua_client.py", line 152, in _call_callback
    self._callbackmap[request_id].set_result(body)
KeyError: 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/asyncua/client/ua_client.py", line 79, in _process_received_data
    self._process_received_message(msg)
  File "/usr/local/lib/python3.7/dist-packages/asyncua/client/ua_client.py", line 98, in _process_received_message
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/usr/local/lib/python3.7/dist-packages/asyncua/client/ua_client.py", line 155, in _call_callback
    f"No request found for request id: {request_id}, pending are {self._callbackmap.keys()}"
asyncua.ua.uaerrors._base.UaError: No request found for request id: 0, pending are dict_keys([14123, 14124])
```

Any help would be much appreciated

Regards,
Kunal

closed 2023-03-28 by schroeder-