FreeOpcUa/opcua-asyncio #58: Question: Client, production ready?

mariopoeta, 2019-06-05:
Hi,

Do you guys think that the client is production ready? :)

oroulet, 2019-06-05:
nobody knows... I am sure it is not worse than some creepy software that people write or use in production everyday but nobody will do support for you so you need to be able to debug it.
I do not think we have any known bugs in client but not so many people use it yet

mariopoeta, 2019-06-05:
I am looking to use the client with AWS Greengrass since now supports python3.7.

I am struggling with the opcua nodejs :)

oroulet, 2019-06-05:
Nodejs has been here a long time. What kind of issues do you have?

mariopoeta, 2019-06-05:
Example, the issue that i am currently troubleshooting.

                const payload = {
                    id: nodeId,
                    value: dataValue.value,
                    timestamp: dataValue.sourceTimestamp,
                };

i want the sourceTimestamp, but i am getting the default value null. but i am able to see sourceTimestamp through quick client or uaexpert.

oroulet, 2019-06-05:
I think which timestamp is sent is an option in a read request

closed 2019-06-18 by mariopoeta