FreeOpcUa/opcua-asyncio #59: Accept SecurityTokens for some time after they expire in SecureConnection

cbergmiller, 2019-06-06:
At the moment `SecureConnection._check_incoming_chunk` checks if the `SecurityToken` that was received along a message is the current token or if it is a valid previous token.
If it is an previous token than all even older tokens are removed and thus declared invalid (if there are any). I don't think this behaviour conforms to the Specification. Tokens could be removed while they still should be considered valid.

From the Specification:

> SecurityTokens have a finite lifetime negotiated with this Service. However, differences between the system clocks on different machines and network latencies mean that valid Messages could arrive after the token has expired. To prevent valid Messages from being discarded, the applications should do the following:
> * Clients should request a new SecurityToken after 75 % of its lifetime has elapsed. This should ensure that Clients will receive the new SecurityToken before the old one actually expires.
> * Servers shall use the existing SecurityToken to secure outgoing Messages until the SecurityToken expires or the Server receives a Message secured with a new SecurityToken. This should ensure that Clients do not reject Messages secured with the new SecurityToken that arrive before the Client receives the new SecurityToken.
> * Clients should accept Messages secured by an expired SecurityToken for up to 25 % of the token lifetime. This should ensure that Messages sent by the Server before the token expired are not rejected because of network delays.

I think we should refactor `SecureConnection` and implement the 25 % token lifetime check, after which the token is removed from the `_old_tokens` list.
Invalid tokens are currently ignored and only a warning is logged. This may pose a security risk? At some point in the past an uaError was raised if an invalid token was received. Did this cause too much trouble? Maybe we should make the behaviour configurable (log warning or raise uaError).

oroulet, 2019-06-06:
if spec says we should accept token that are 25% over the lifetime, then we should do it.

cbergmiller, 2019-06-07:
In part 4 the spec talks about _rejecting_ messages with expired security token ids. @oroulet do you know how that would be done? By sending a `ErrorMessage` with `BadSecureChannelTokenUnknown`?

In part 6 6.7.6 the spec talks abour _reporting_ an error:
> If the Message is secured with symmetric algorithms then a Bad_SecureChannelTokenUnknown error shall be reported if the TokenId refers to a SecurityToken that has expired or is not recognized.

By reporting it to the application or to the server?

Or should we not bother with rejection and either ignore wrong token ids (current implementation) or close the session?

oroulet, 2019-06-07:
When we started that project our goal was to make it work, I mean make it able to communicate with aas many server/Client as possible. so we never rejected anything suspect.
But now things have stabilized and maybe we should start being more picky. As far as I understandm the sentence above is for the server. On client side we should close connection,  raise an exception if we can and cancel all waiting futures... of we just accept it