FreeOpcUa/opcua-asyncio #60: Server datavalue history doesn't save data in sqlite3 database

okyame, 2019-06-14:


okyame, 2019-06-17:
I find that, in history.py, in class `SubHandle` (line 185)  ,`datachange_notification` function is never called.

oroulet, 2019-06-17:
if you add a print there, it is never printed? it is supposed to be tests for that method..

okyame, 2019-06-17:
No, it's never printed.

okyame, 2019-06-17:
When I stop the server with 'KeyboardInterrupt' shortcut, it calls `datachange_notification` before it stopped. But nothing is saved in database.

```
Node(NumericNodeId(ns=2;i=2))
Endpoints other than open requested but private key and certificate are not set.
Executing <Task pending coro=<main() running at examples/server-datavalue-history-asyncio.py:39> wait_for=<_GatheringFutur
e pending cb=[<TaskWakeupMethWrapper object at 0x7f40efd24858>()] created at /usr/lib/python3.6/asyncio/tasks.py:549> cb=[
_run_until_complete_cb() at /usr/lib/python3.6/asyncio/base_events.py:185] created at /usr/lib/python3.6/asyncio/base_even
ts.py:452> took 0.736 seconds
0.1
0.2
0.30000000000000004
^C
Executing <Handle <TaskWakeupMethWrapper object at 0x7f40efd246a8>(<Future finis...events.py:284>) created at /usr/lib/python3.6/asyncio/futures.py:375> took 3.472 seconds
Enter in subhander class !
Enter in subhander class !
Enter in subhander class !
Enter in subhander class !
Enter in save_node_value class !
Enter in save_node_value class !
Enter in save_node_value class !
Enter in save_node_value class !
Traceback (most recent call last):
  File "examples/server-datavalue-history-asyncio.py", line 60, in <module>
    loop.run_until_complete(main())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 460, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 427, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1432, in _run_once
    handle._run()
  File "/usr/lib/python3.6/asyncio/events.py", line 145, in _run
    self._callback(*self._args)
  File "examples/server-datavalue-history-asyncio.py", line 47, in main
    time.sleep(1)
KeyboardInterrupt
```

okyame, 2019-06-17:
Finally, i found and fixed the issue: ""conn_commit()" is never called.

I modify this function in the `history_sql.py ` file like this:

```
    async def _execute_sql(self, sql: str, params: Iterable = None):
        await self._loop.run_in_executor(None, self._conn.execute, sql, params or ())
        await self._loop.run_in_executor(None, self._conn.commit)
```

oroulet, 2019-06-17:
can you make a PR? that is easy

okyame, 2019-06-17:
Here is PR

cbergmiller, 2019-09-01:
Fixed by #65

closed 2019-09-01 by cbergmiller