FreeOpcUa/opcua-asyncio #61: Await datachange?

luxderfux, 2019-06-16:
Is there a possibility to create a subscription and await data changes (in an endless loop) instead of triggering the `datachange_notification` method in a subscription handler class?

oroulet, 2019-06-16:
I do not think so. But it is a very very good point you are raising. @cbergmiller

cbergmiller, 2019-06-17:
We don't have this kind of API at the moment. You can implement it in your code with [asyncio.Queue](https://docs.python.org/3/library/asyncio-queue.html#queue). Use `queue.put_nowait()` inside the `datachange_notification` method and await new data with `queue.get()` inside an async task.

oroulet, 2019-06-17:
@cbergmiller i am wondering how such an API would be... And should we internally simply use a queue. The re is some thinking needed here

cbergmiller, 2019-06-17:
I already have thought about this (replacing callbacks with queues). This could be an alternative API for subscribing to data changes. I don't wont to remove the callback API because it sometimes is exactly what is needed. With only a queue as API you would be forced to always use long running tasks even if you do not need to await the data changes.

As for the API design, the `Subscription` class could expose a `Queue` instance property that is used to send changes if there was no handler passed in `Client.create_subscription()`.

luxderfux, 2019-06-17:
Thanks! The idea with the asyncio.Queue sounds promising.
I will implement that solution and post the code here later on... maybe it's useful for other users.

luxderfux, 2019-06-19:
Ok it was pretty easy to implement:

```python
import asyncio
from asyncua import Client, Node, ua

queue = asyncio.Queue()

class sub_handler:
    def datachange_notification(self, node: Node, val, data):
        timestamp_ns = int(data.monitored_item.Value.SourceTimestamp.timestamp() * 1e9)
        queue.put_nowait({'timestamp': timestamp_ns, 'value': val})

async def main():
    async with Client(url='opc.tcp://localhost:4840/') as client:
        var          = await client.get_objects_node().get_child(["data"])
        handler      = sub_handler()
        subscription = await client.create_subscription(500, handler)
        await subscription.subscribe_data_change(var)

        while True:
            data = await queue.get()
            print(data)

if __name__ == '__main__':
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
    loop.close()
```

I have two concerns / questions with that solution:
1. Thread-safety
`asyncio.Queue` is not thread-safe. I guess that the `datachange_notification`callback function is invoked by a different thread!? Are we running into troubles then?
2. Efficiency
Sometimes I handle quite huge arrays (>10.000 samples). Will introduce `queue.put_nowait(...)` and `queue.get()` additional copies? I tried to benchmark it but couldn't measure any overhead of the queue...

cbergmiller, 2019-06-19:
Thanks for sharing your implementation.
Regarding your concerns:
1. `datachange_notification` is called by the main event loop (inside an asyncio socket callback). So there is no need for thread safety.
2. My understanding of `asyncio.Queue` is that no copies of the items are generated. Once the item is consumed by `get` it also should not be referenced by the queue any longer (otherwise it would be blocking garbage collection).

I am not entirely sure if it is ok to ignore [task_done](https://docs.python.org/3/library/asyncio-queue.html#asyncio.Queue.task_done). It probably only keeps count of the unconsumed items and therefore optionally.

luxderfux, 2019-06-19:
Perfekt, thanks for clarification! Although it seems works fine at the moment, I added the `queue.task_done()` - also in the example above.

closed 2020-12-15 by swamper123