FreeOpcUa/opcua-asyncio #64: Next planned release?

peteryin21, 2019-06-20:
Hi, I am in the process of converting our codebase from `python-opcua`, and we rely on a feature to be able to pass in an event loop into the server from [this commit](https://github.com/FreeOpcUa/opcua-asyncio/commit/56cf68dfc51cb80418ed7a94e6b84edac2ffaabe#diff-85f90aee7fd78d5c077634c3b26eb9bd). 

I noticed this is not in 0.5.0. When is the next planned release?

oroulet, 2019-06-21:
OK made a release just now since we have a few bigger changes in the pipeline

peteryin21, 2019-06-21:
@oroulet Thanks! For future reference, is there a forum/email chain where new features and releases are discussed?

oroulet, 2019-06-21:
We just discuss things in pull requests and bug reports.

closed 2019-06-21 by peteryin21