FreeOpcUa/opcua-asyncio #68: rename get_value to read and set_value to write?

oroulet, 2019-06-29:
read and write are more explicit and are the words used by spec. so it would be better. I always regretted choosing get and set originally, but that is an incompatible API change that might make some users not really happing
@cbergmiller

afcmrp, 2019-12-20:
Why not rename them and then set get_value = read and set_value = write in class Node to provide backwards compatibility? I agree that it is better to follow the naming convention in the spec and a rename change only gets worse the longer time it takes.

oroulet, 2019-12-20:
Good point. PR welcome. In fact you can just change the entire library with find and sed in 2 min

oroulet, 2019-12-20:
well so easy that I just did it

swamper123, 2020-09-14:
This was done with #120 so I guess it can be closed.

closed 2020-09-14 by swamper123