FreeOpcUa/opcua-asyncio #70: Client reconnect behaviour?

dannmartens, 2019-07-12:
Hi,

We see a warning printed by our client, whenever our OPC UA server restarts.

[WARNING] asyncua.client.ua_client.UASocketProtocol - ServiceFault from server received  in response to PublishRequest

Once this happens, the client's data change subscriptions no longer work.

Is there any particular configuration needed in code to make the client reconnect and resubscribe again? Can we get notifications from these ServiceFaults to deal with it ourselves?

Thanks for your feedback,
 Dann

dannmartens, 2019-07-22:
I solved this by adding a watchdog task to the asyncio loop: if a server value doesn't change N consecutive times, the client resubscribes.

closed 2019-07-22 by dannmartens

oroulet, 2019-07-22:
Yes there is not high level automatic reconnection. You need to make your own

f3lixTRS, 2022-07-29:
> I solved this by adding a watchdog task to the asyncio loop: if a server value doesn't change N consecutive times, the client resubscribes.

Hello, can you post the code you used for fix the problem? I'm experiencing the same problem and I don't know how to fix it.

dannmartens, 2022-07-29:
@f3lixTRS I've taken a quick look at how I did this:

1) Add a "watchdog" task to the loop which can signal that it is time to resubscribe:
```
self.loop.create_task(self._watchdog())
```
This needs to be done before the first subscription attempt. After that you can enter a "subscribe loop":
```
    while self.resubscribe:
        handler = NotificationHandler(self._callback)
        try:
            self.loop.run_until_complete(self.subscriber.subscribe(handler, self))
            if self.resubscribe:
                self.loop.run_until_complete(self._breathe()) # Breathe just waits 3 seconds
        except:
            _logger.exception("The subscription loop failed!")
```

2) The watchdog then ticks on its own, every second, until N (here 3 seconds) is reached:
```
    async def _watchdog(self):
        while self.resubscribe:
            elapsed = time.perf_counter() - self.perf_counter
            if self.watchdog_on and elapsed > 3:
                self.watchdog_on = False
                _logger.error('Watchdog timer exceeded threshold: cancelling OPC UA subscription!')
                self.subscriber.cancel_subscription()
            await asyncio.sleep(1)
```
3) Update the last noftification time in the handler, so the watchdog will know.
```
    def _callback(self, value):
        """Callback for NotificationHandler"""
        self.perf_counter = time.perf_counter()
```

f3lixTRS, 2022-07-29:
Thanks a lot @dannmartens for your quick response. 
Now I will try to figure out how to put this in my code (I'm new in python and in asyncua).

Thanks again.