FreeOpcUa/opcua-asyncio #71: Datachange not acknowledged in History

Nanahara, 2019-07-16:
Hello,

I try to test history function and I use `server-datavalue-history.py` from #63 . I see in log information that the datachange notifications are never acknowledged: see attached file: 
[log.txt](https://github.com/FreeOpcUa/opcua-asyncio/files/3397317/log.txt)

We see that `AvailableSequenceNumbers` list is growing up indefinitely with opcua-asyncio  whereas a publish request with acks is generated with python-opcua

Have you an idea about this issue ?

N.B: I run server example without connected client.

Nanahara, 2019-07-24:
In Subscription class (subscription.py) following lines have been removed in publish_callback function in PR #45 

`ack = ua.SubscriptionAcknowledgement()`
 `ack.SubscriptionId = self.subscription_id`
 `ack.SequenceNumber = publishresult.NotificationMessage.SequenceNumber`
 `await self.server.publish([ack])`

If I restored this code, the datachange notifications from history subscription are correctly acknowledged.

What was the initial purpose of removing the `server.publish` call in publish_callback ?

cbergmiller, 2019-07-25:
Hi, thanks for investigating this issue!
The goal of #45 was to further decouple `Subscrition`, `UaClient` and `InternalSession`.
In case of the client, `UaClient._publish_loop` now takes care of sending the `SubscriptionAcknowledgement`.
I guess i didn't fully understand the importance of publishing the Ack on the server side.
This has probably to be done somewhere in `InternalSubscription`.
I will have to take some time to figure this out. A good first step would be to write a test to guard against a regression in the future.

commit e487f7b9 (2019-09-02): Fix #71 (missing publish call in internal server subscriptions) (#80)

* add missing publish call

* added InternalSubscription is_for_client flag

* rename is_for_client to na_acks, use isession.external prop

* docs
closed 2019-09-02 by oroulet