FreeOpcUa/opcua-asyncio #72: Subscription mechanism and Publish on binary HTTPS server ?

Nanahara, 2019-07-18:
Hi,

I'm trying to implement binary/https opc-ua server by using Tornado https server. But I don't know how subscription mechanism should be processed over HTTPS that is a stateless protocol. 
How is processed a publish request ? How is sent a publish response ? (Maybe Websocket ?)

Have you an idea about this subject ?

oroulet, 2019-08-14:
No. idea. must be written in spec somewhere. What are you trying to do? sounds like a huge project to reimplement everything on tornoda. You should better implment https protocol in opcua-asyncio. by generating xml based format and using a small http server like aiohttp

Nanahara, 2019-08-22:
Hi,

The official spec is not very detailed about https usage and I have not found very resources or project about this subject, I will continue to investigate it.
Concerning tornado usage, I have no choice to use it. My project is to implement opc ua in an existing web application on tornado.

For moment I have a prototype using tornado and opcua-asyncio. I implemented an handler to reply post request by using for example uaprocessor.py and use UaExpert from Unified Automation to test it.

I will publish my prototype on github if you are interesting when i have a clean version.