FreeOpcUa/opcua-asyncio #77: SourceTimestamp and ServerTimestamp created as naive Datetime?

oisene, 2019-08-07:
In opcua.ua.uatypes at line 21 we have:

`FILETIME_EPOCH_AS_DATETIME = datetime(1601, 1, 1)`

I am happy to be corrected, but to me this creates what the Python documentation calls a naive datetime, that is, without time zone information, and this will return the time local to my time zone. 

In other words, these return different results; shouldn't the last one  be used?

```
datetime(1601,1,1).timestamp()
datetime(1601,1,1,tzinfo=timezone.utc).timestamp()
```

(All timestamps in the OPC UA standard are in UTC, AFAIK)

oroulet, 2019-08-07:
You are probably correct. When I wrote that code, there were many issues with time zones in python. Maybe it works now. Send a PR

oisene, 2019-08-07:
Thanks for quick response; I will after some testing.

oisene, 2019-08-08:
Changing it to an "aware" Datetime breaks a lot of the tests. Also using the official timestamp for 1601-01-01 (-11644473600) breaks quite a few tests, looks there are still many issues with datetime in Python for dates pre-Gregorian calendar. In my own tests I get errors when converting between UTC and local time for the date mentioned above.

pbertoni89, 2021-12-10:
Hi there! Are there some news on that? All server timestamps are UTC and I'll be glad to set another timezone once, instead of forcing clients to make assumptions and to adjust them from the client side

oroulet, 2021-12-10:
I have not heard anything so I suppose the state is the same. you are welcome to give it a shot @pbertoni89

AndreasHeine, 2021-12-10:
in datetime you can set "tzinfo" (https://docs.python.org/3/library/datetime.html#datetime.datetime.tzinfo) if it is "None" its UTC anyways.

oroulet, 2021-12-10:
looking at code I see that we call datetime.utcnnow() everywhere so I think the timezone is set in fact. Does that issue really exist @pbertoni89 ?

pbertoni89, 2021-12-10:
Well I'm not very proficient in the library itself (I'm a novice both in OPCUA and ASYNCIO), however I see

```shell
$ grep -rIn timezone | grep -v tests
```

> asyncua/common/xmlparser.py:263:            # According to specs, DateTime should be either UTC or with a timezone.
> asyncua/ua/uatypes.py:13:from datetime import datetime, timedelta, MAXYEAR, timezone
> asyncua/ua/uatypes.py:45:FILETIME_EPOCH_AS_UTC_DATETIME = FILETIME_EPOCH_AS_DATETIME.replace(tzinfo=timezone.utc)
> asyncua/server/event_generator.py:95:            self.event.LocalTime.Offset = -(time.altzone if localtime.tm_isdst else time.timezone)

So my question was incorrectly formed, in facts I'd just localize somehow a running server. I apologize for any misunderstanding

closed 2023-03-29 by schroeder-