FreeOpcUa/opcua-asyncio #79: Memory increases monotonically from repetitive client connects and disconnects.

dannmartens, 2019-08-27:
We've observed process memory ballooning to several GBs over time, starting from an initial 80 MB, running an asyncio OPC UA server.

The process is very active doing other things as well, but as long as no OPC UA clients connect, memory stays constant. 

A burn-in test app was created to perform a continuous connect, write value, disconnect cycle and we can observe the memory footprint increasing slowly. 

I ran mem_top against the server with the following results:
`refs:
18566   <class 'dict'> {1: PublishResult(SubscriptionId:78, AvailableSequenceNumbers:[1], MoreNotifications:False, Notifica
10602   <class 'dict'> {'__module__': 'asyncua.ua.object_ids', 'Null': 0, 'Boolean': 1, 'SByte': 2, 'Byte': 3, 'Int16': 4,
9282    <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
9281    <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
9280    <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
9279    <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
9278    <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
9277    <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
9276    <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
9275    <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,

bytes:
295016   {'__module__': 'asyncua.ua.object_ids', 'Null': 0, 'Boolean': 1, 'SByte': 2, 'Byte': 3, 'Int16': 4,
295016   {1: PublishResult(SubscriptionId:78, AvailableSequenceNumbers:[1], MoreNotifications:False, Notifica
83656    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
83648    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
83640    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
83624    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
83616    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
83608    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
83600    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
83592    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,

types:
244138   <class 'dict'>
61728    <class 'list'>
57277    <class 'asyncua.ua.uatypes.StatusCode'>
57277    <class 'asyncua.ua.uatypes.Variant'>
57276    <class 'asyncua.ua.uatypes.DataValue'>
34849    <class 'asyncua.server.address_space.AttributeValue'>
22430    <class 'asyncua.ua.uaprotocol_auto.MonitoredItemNotification'>
22356    <class 'tuple'>
12341    <class 'asyncua.ua.uatypes.NumericNodeId'>
11588    <class 'asyncua.ua.uatypes.NodeId'>
`

We're still investigating this, but would like some feedback in case others have noticed similar behaviour.

dannmartens, 2019-08-27:
For the synchronous variant, I found https://github.com/FreeOpcUa/python-opcua/issues/744

oroulet, 2019-08-27:
Thanks for looking at these things!
What are you clients doing? subscribing? or just reading?

dannmartens, 2019-08-27:
The "production" WinCC OA client does write to a limited set of nodes, and is subscribed to a few others.

The burn-in client just writes a string every three seconds to a single node. It's written with the asyncua client library.  
`async with Client(url=projector.config.get('OPCUA_URL')) as client:`

The impression we get is that the more intense the communication over OPC UA, the more memory is allocated. We're running the same server multiple times one next to another, and we can see large differences between those left quiet and those involved in heavy communication. 

In a separate experiment, I bypassed the event logic in the server which handles the value change of the string write, and let it do its work as if an external client had written a value. In that experiment, the memory footprint remains stable.

oroulet, 2019-08-27:
I cannot imagine how writting to a node can have a memory leak. The code is simple. The session opening stuff might have( there has been one) but my first suspect would be the subscription mechanisms

cbergmiller, 2019-08-27:
It may be related to https://github.com/FreeOpcUa/opcua-asyncio/issues/71 (a bug introduced by my changes in https://github.com/FreeOpcUa/opcua-asyncio/pull/45).

dannmartens, 2019-08-27:
The list of possible culprits by mem_top shows these weird arrays of integers. There is some rule in the madness, as well: the refs increment from 9275 to 9282 in sequence and the corresponding byte sizes increment, as well. 

I have no idea what those are. Could these be those sequence numbers #71 mentions?

oroulet, 2019-08-27:
yes I think these are the acknowledgments. try to track them and see if they get removed

dannmartens, 2019-08-27:
The refs count keeps rising for those arrays, it appears. This is a few hours later:
`refs:
20478   <class 'dict'> {1: PublishResult(SubscriptionId:78, AvailableSequenceNumbers:[1], MoreNotifications:False, Notifica
10602   <class 'dict'> {'__module__': 'asyncua.ua.object_ids', 'Null': 0, 'Boolean': 1, 'SByte': 2, 'Byte': 3, 'Int16': 4,
10239   <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
10238   <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
10237   <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
10236   <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
10235   <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
10234   <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
10233   <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
10232   <class 'list'> [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,

bytes:
295016   {'__module__': 'asyncua.ua.object_ids', 'Null': 0, 'Boolean': 1, 'SByte': 2, 'Byte': 3, 'Int16': 4,
295016   {1: PublishResult(SubscriptionId:78, AvailableSequenceNumbers:[1], MoreNotifications:False, Notifica
92264    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
92256    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
92248    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
92240    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
92232    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
92224    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
92216    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,
92208    [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27,

types:
255152   <class 'dict'>
67464    <class 'list'>
59996    <class 'asyncua.ua.uatypes.Variant'>
59995    <class 'asyncua.ua.uatypes.DataValue'>
59995    <class 'asyncua.ua.uatypes.StatusCode'>
34849    <class 'asyncua.server.address_space.AttributeValue'>
25149    <class 'asyncua.ua.uaprotocol_auto.MonitoredItemNotification'>
24936    <class 'tuple'>
12341    <class 'asyncua.ua.uatypes.NumericNodeId'>
11584    <class 'asyncua.ua.uatypes.NodeId'>
`

oroulet, 2019-08-27:
@cbergmiller are you looking at that stuff? it seems a bit hard to fix..
My current understanding is that the you moved some of the functionnaly to a publish_loop method which only exist in UaClient. It is missing in InternalSession. In order not to repeat code I think that the code should be moved to the Susbcription class or file and only started once in InternalSession(Server) and once in UaClient.
Am I correct?

@dannmartens this seems to be a bug related to server side subcsription. I am really not sure this is the same as what you have. Do you use subscription server side?

cbergmiller, 2019-08-27:
@oroulet i am not sure of InternalSession needing a publish loop. I think only the acknowledgement of (server internal) subscription updates is what is missing in InternalSession. It is indeed very hard to understand in detail. I will take a look soon and post an update in #71.

oroulet, 2019-08-27:
I confirm, it is hard 🤯. And yes the missing stuff is the acknowledgement from server side. Also the equivalent of uaclient is internal session. It is the only place we know this is server side. But I cannot find where publish () is called server side. It most be called otherwise we miss some publish requests

dannmartens, 2019-08-27:
@oroulet Yes, there is an internal subscription set up, in order to respond to external client writes.

```python
async def subscribe(self, vars):
     handler = SubscriptionHandler()
     subscription = await self.server.aio_obj.create_subscription(100, handler)
     handles = await subscription.subscribe_data_change(vars)
     _logger.info('Subscribed to internal variables.')
```

This just happens once in the entire lifecycle of the server, at startup.

The WinCC OA clients do their own subscribes, but the burn-in client doesn't do any at all: it just writes to that var the internal logic is subscribed to.

cbergmiller, 2019-08-27:
@oroulet `self.server.publish([ack])` was called at the end of  `Subscription.publish_callback` prior to #45.
`self.server` being either `InternalSubscription` or `UAClient`.
I think it publish may have to be called at the end of `InternalSubscription.publish_results` i will try to test this.
It may be that `InternalSubscription._not_acknowledged_results` is the dict that keeps growing because of this bug.

dannmartens, 2019-08-28:
Some more statistics collected: the first twelve servers are "in rest," they generate a lot of Modbus traffic on a 500 msec cycle, but no OPC UA clients are connected. The last one is running in a staging environment with a WinCC OA OPC UA client writing every 10 seconds (it has subscribed, as well).

CPU %               MEM USAGE / LIMIT    MEM %               NET I/O

2.42%               75.28MiB / 7.79GiB   0.94%               595MB / 800MB
1.81%               73.73MiB / 7.79GiB   0.92%               592MB / 797MB
2.39%               74.8MiB / 7.79GiB    0.94%               592MB / 798MB
4.05%               86.33MiB / 7.79GiB   1.08%               852MB / 1.15GB
3.14%               81.82MiB / 7.79GiB   1.03%               861MB / 1.17GB
1.27%               73.79MiB / 7.79GiB   0.93%               330MB / 385MB
1.47%               74.05MiB / 7.79GiB   0.93%               3.1GB / 4.17GB
2.82%               74.24MiB / 7.79GiB   0.93%               3.1GB / 4.17G
2.86%               74.28MiB / 7.79GiB   0.93%               2.87GB / 3.87GB
3.77%               81.74MiB / 7.79GiB   1.02%               3.19GB / 4.34GB
2.46%               82.2MiB / 7.79GiB    1.03%               3.19GB / 4.35GB
1.42%               73.66MiB / 7.79GiB   0.92%               290MB / 333MB

3.72%               1.019GiB / 7.79GiB   13.09%              171MB / 229MB

Modbus traffic can be accounted for most of the NET I/O, but even after 24 hours, the test server tops out at just 171 MB. As opposed to the "quiet" servers—one group already running for 5 days, the second group three weeks—the test server's memory usage is now over 1 GB. The other servers have been occasionally interfaced with, or not at all and their memory usage seems to reflect that.

oroulet, 2019-10-25:
@dannmartens is that fixed?

dannmartens, 2019-10-25:
It's spinning in production, right now: memory allocation is stable!

oroulet, 2019-10-25:
great to hear!

closed 2019-10-25 by oroulet