FreeOpcUa/opcua-asyncio #82: Certificate handling

mitchpi, 2019-09-23:
Hi Guys, thanks for a great package!

I'm struggling a little with the certificate handling. The server always accepts self signed certificates, shouldn't the certificates have to be moved to a trust list prior to it being trusted? And hence the client will be rejected and not able to view the server nodes?

Cheers. Mitch.

oroulet, 2019-09-23:
It should... Patch welcome ;-)

mitchpi, 2019-09-23:
Thanks for the prompt response! I'll investigate, may be a bit beyond me.

JoeyFaulkner, 2020-05-06:
Hi there! 

also echo mitchpi's sentiment: this is an awesome package! I would like to work on this certificate handling. Do you have any tips on where this needs to be implemented and if there's anything I need to be careful of?

JoeyFaulkner, 2020-05-11:
@oroulet i think we can close this issue now because of the closed PR?

closed 2020-05-11 by oroulet