FreeOpcUa/opcua-asyncio #85: MaxBrowseContinuationPoints null value breaks B&R client compatibility

kohtala, 2019-10-02:
I got second hand report that B&R OPC-UA client (of unknown version) was unable to connect to opcua-asyncio server. UA_Connect() fails with Bad_TypeMismatch 0x80740000.

The reported error was on MaxBrowseContinuationPoints that returns Variant Type Null as the value instead of the the UInt16 that was expected. The specification says "The value 0 indicates that the Server does not restrict the number of parallel continuation points the client should use." and I believe the value should be set to this instead of leaving it at null.

It was confirmed, that setting the value to 0 fixes the issue.

There are also other mandatory variables without value, but apparently the client did not read them or error on them.

kohtala, 2020-06-25:
Hi.

If you get the error, would you tell the version of B&R OPC-UA client you are using?

Setting the value can be done with something like (adapt to your need)

    server = Server()
    objectentity = server.get_objects_node()
    servervars = await objectentity.get_child("Server")
    ob = await servervars.get_child("ServerCapabilities")
    maxbrowsenode = await ob.get_child("MaxBrowseContinuationPoints")
    await maxbrowsenode.set_value(0, ua.VariantType.UInt16)

scholli94, 2020-06-26:
Hey kohtala, i tried to implement your fix now for some hours but it wont work for me :-(
Right now i am using the following code:

import asyncio
import copy
import logging
from datetime import datetime
import time
from math import sin
import sys
sys.path.insert(0, "..")

from asyncua import ua, uamethod, Server
                                            #Serveradresse festlegen
server = Server()
url = 'opc.tcp://192.168.0.8:4841'
server.set_endpoint(url)
name="Testserver"

objectentity = server.get_objects_node()
loop=asyncio.get_event_loop()
async def main():
    servervars =await objectentity.get_child("Server")
    ob = await servervars.get_child("ServerCapabilities")
    maxbrowsenode = await ob.get_child("MaxBrowseContinuationPoints")
    await maxbrowsenode.set_value(0, ua.VariantType.UInt16)
loop.run_until_complete(main())

server.start()                                  #Server starten
print("Server started at {}".format(url))

while True:
    print("test!")
    time.sleep(2)

and the following is the error code:

Traceback (most recent call last):
  File "C:/Users/Scholli/Dropbox/opcua/opcua.py", line 25, in <module>
    loop.run_until_complete(main())
  File "C:\Users\Scholli\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:/Users/Scholli/Dropbox/opcua/opcua.py", line 21, in main
    servervars =await objectentity.get_child("Server")
  File "C:\Users\Scholli\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asyncua\common\node.py", line 483, in get_child
    result.StatusCode.check()
  File "C:\Users\Scholli\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asyncua\ua\uatypes.py", line 223, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadNodeIdInvalid: "The syntax of the node id is not valid."(BadNodeIdInvalid)

Process finished with exit code 1

Since i have no further ideas how to fix it i am coming back here to ask you once again 

Thanks a lot.
Regards Felix

closed 2023-03-28 by schroeder-