FreeOpcUa/opcua-asyncio #93: `_trigger_publish` Question

chrisjbremner, 2019-10-17:
@oroulet I am trying to diagnose why I am seeing a latency (~0.2s, even when I decrease the publishing interval in the client to 1ms) between when I trigger an event and when the event is received on the client side. I suspect that this is probably an acceptable latency (especially because I am running the server on a machine with low computational power), but I was digging through the server code to see if there's anything that could be done to try and drop this latency further, and I came across this section of code which appears to me to not be reached. I created this issue to ask the question about whether I am misreading it, or if it's a typo that could potentially help me trigger events faster. If it is something that should be changed, I can make a PR.

The section of code I am looking at is [here](https://github.com/FreeOpcUa/opcua-asyncio/blob/08861799d3413345dcc5b4ae387427739b864bad/asyncua/server/internal_subscription.py#L60-L66). In order to enter this if statement, the subscription loop task must be running, and it is started [here](https://github.com/FreeOpcUa/opcua-asyncio/blob/08861799d3413345dcc5b4ae387427739b864bad/asyncua/server/internal_subscription.py#L50-L51). In order to start the subscription loop task, the `RevisedPublishingInterval` must be greater than 0.0. Therefore, it seems impossible that both the subscription loop task is running and the `RevisedPublishingInterval` can be less than or equal to 0.0, so `publish_results()` is not called from within `_trigger_publish`. Is it possible it is supposed to be something like this instead? (added a `not`):
```python
if not self._task and self.data.RevisedPublishingInterval <= 0.0:
```

cbergmiller, 2019-10-18:
Hi!

The Spec. says:

> requestedPublishingInterval - Duration
> This interval defines the cyclic rate that the Subscription is being requested to return Notifications to the Client. This interval is expressed in milliseconds. This interval is represented by the publishing timer in the Subscription state table (see 5.13.1.2). The negotiated value for this parameter returned in the response is used as the default sampling interval for MonitoredItems assigned to this Subscription. If the requested value is 0 or negative, the server shall revise with the fastest supported publishing interval. 

This seems to be indeed a bug in `InternalSubscription`. Have you tried to run the serer with the added `not`?

chrisjbremner, 2019-10-18:
Hi @cbergmiller, thanks for checking the spec! I wanted a sanity check before making the change, but I just tested it out and it appears to work, so I made https://github.com/FreeOpcUa/opcua-asyncio/pull/94

closed 2019-10-25 by oroulet