FreeOpcUa/opcua-asyncio #96: Questtion about asyncio usage for node management

Nanahara, 2019-10-24:
Hello,

After having studied this library (very good job by the way), I don't understand why coroutines (with asyncio) are used on node access and node management functions. Indeed, these functions read or write on address space that is loaded in RAM and I think using coroutines increases latency when I create new nodes or search a node.

To test my assumption I removed async declaration from node functions and I created 30 objects with 100 variables and I saw a significant latency difference between asyncio usage and non asyncio usage.

Is there any particular reason to have used coroutine that I have missed ?

oroulet, 2019-10-24:
Thanks for the comments. I suppose you are talking about the server side API. The reason is that python-opcua had the exact same API for client and server. We ported the API to asyncio, thus client and server have async API (Node class for example) although some methods could be sync. We discussed this earlier but did not find any nice solutions without repeating a lot of code betweeen client and server
everything server speicifc should be sync ...
Any suggestions welcome..

Nanahara, 2019-10-30:
Thanks for your answer. If I have an idea about this subject I will share here.

oroulet, 2019-10-30:
Also we already created a server specific method to write quickly a  value to a variable. We can add one to create nodes efficiently if there is a need. Patch welcome