FreeOpcUa/opcua-asyncio #100: Can't access attributes and values from nodes

SaschaSchlemmer, 2019-11-07:
I have a local opc ua server running (https://www.unified-automation.com/products/server-sdk/c-ua-server-sdk.html) and I can connect to it via 

```python
from asyncua import Client

client = Client(url='opc.tcp://XXXX:48010')
await client.connect()
```

then I navigate to the root node via 
```python
root = client.get_root_node()
```

when I try to get any attribute/value however e.g. via `await root.get_children()` I get an `AttributeError`

```
~\AppData\Local\Continuum\anaconda3\envs\py37\lib\site-packages\asyncua\client\ua_client.py in _send_request(self, request, timeout, message_type)
    122         self._callbackmap[self._request_id] = future
    123         msg = self._connection.message_to_binary(binreq, message_type=message_type, request_id=self._request_id)
--> 124         self.transport.write(msg)
    125         return future
    126 

AttributeError: 'NoneType' object has no attribute 'write'.

I also successfully tried another node on the demo server besides the root node and the same issue occurs. Can anybody point me to what I am doing wrong?
```

RychenerLorenz, 2019-11-07:
Hi there, 

I just tried to re-build the things in your code and it works perfectly fine. Could you provide more context?

oroulet, 2019-11-07:
From the error it looks like you are not connected. Make a sample code and paste it here

swamper123, 2021-01-14:
Since this issue seems to be orphand or solved by itself - Case closed

closed 2021-01-14 by swamper123