FreeOpcUa/opcua-asyncio #103: examples not working

Armagetron, 2019-11-25:
As discussed in #1 I have problems with some of the examples. Most of them are not updated for asyncio. E.g. `server-custom-object.py` fails with
```
Traceback (most recent call last):
  File "./server-custom-object.py", line 42, in <module>
    mycustomobj_type.add_variable(0, "var_should_be_there_after_instantiate", 1.0) # demonstrates instantiate
AttributeError: 'coroutine' object has no attribute 'add_variable'
sys:1: RuntimeWarning: coroutine 'Node.add_object' was never awaited
sys:1: RuntimeWarning: coroutine 'Node.get_child' was never awaited
sys:1: RuntimeWarning: coroutine 'Node.add_object_type' was never awaited
sys:1: RuntimeWarning: coroutine 'Server.register_namespace' was never awaited
```

I used a simple environment in a Docker container like this one:
```
FROM ubuntu:18.04

RUN apt-get update && apt-get install -y python3 python3-pip git
RUN pip3 install asyncua
RUN git clone https://github.com/FreeOpcUa/opcua-asyncio.git
WORKDIR opcua-asyncio/examples 
CMD ["python3", "./server-custom-object.py"]
```
Other examples that I could not get to work are: `server-callback.py`, ` server-create-custom-structures.py`, `server-enum.py`, `server-events-history.py`, `server-instantiate-object.py`, `server-ua-python-mirror.py`, `server-with-encryption.py`, `server-xmlexporter.py`. I have not tested any client examples. Am I using them wrong?

oroulet, 2019-11-25:
As you wrote, some examples are not ported. Help welcome, Should be easy. You can try

tobiasht, 2020-03-03:
I can have a shot at it, but I don't have much experience using async. Could you possibly port "server-create-custom-structures.py" so i can have some more examples to look at?

danmoser, 2022-05-03:
Folks, FYI `server-custom-object.py` is working good as of today.

However, I'm in trouble running `server-custom-structures-before-1.04.py`...
The problem occurs at `server.load_type_definitions()`. It does not recognize the struct defined in the `DataTypeDictionaryBuilder`.

I would like to use `DataTypeDictionaryBuilder` with asyncio. Unfortunately my knowledge doesn't allow me to debug further. Thanks in advance!

danmoser, 2022-05-03:
> server-create-custom-structures.py

The asyncua example `server-custom-structures-amd-enums.py` is working nicely!

schroeder-, 2022-05-03:
I tried `server-custom-structures-before-1.04.py` on current master and it's working for me. DataTypeDictionaryBuilder exists see: https://github.com/FreeOpcUa/opcua-asyncio/blob/84b4a60f39c9d71e4c22b9c036611783d03de13d/asyncua/common/type_dictionary_builder.py#L126 

Still i wouldn't use the old v1.03 stuff, if you not rly need to support old clients, because v1.04 structs offer a lot more features and are more future proof.

oroulet, 2022-05-03:
In addition 1.04 is almost correctly implemented. The protocol was much cleaner. That is not the case of 1.03

closed 2023-03-28 by schroeder-