FreeOpcUa/opcua-asyncio #105: HasNotifier implementation for events

mar-ar, 2019-11-29:
Hi guys,

I realized that the implementation of the events isn´t complete at the moment.
According to the specs each event should be received if you subscribe to the server node.
At the moment there is no way to implement an event hierarchy.
Here is a event hierarchy from the specs:

[image]

I will try to implement such a behavior.

oroulet, 2019-11-29:
There are many things in the spec which is not implemented by any server or clients ;-)
Or server supports events coming from any node. This is already rare (at least it was)
But yes if you find missing stuff, you are welcome to implement them

mar-ar, 2019-11-29:
> There are many things in the spec which is not implemented by any server or clients ;-)

You are right. I will look at it and then decide. If HasNotifier would be a two way reference it would be easier.

aldoshkind, 2025-02-08:
Hello @mar-ar @oroulet !
Can you please tell me if event propagation implemented at the moment? I am working on a project where this feature would be very handy.