FreeOpcUa/opcua-asyncio #106: Running a test OPC-UA server

gkotian, 2019-11-29:
!! N00b alert !!

I'm just getting into the OPC world, and one of the very first things I'm doing is to try and run a test OPC-UA server using this repo.

I created a docker image with the given [Dockerfile](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/Dockerfile) using:
```
docker build -t test-opc-server .
```

Then I opened a container instance with:
```
docker run --rm -it test-opc-server bash
```

Finally I ran `uaserver` within the container, and got this:
```
root@1bd3fc641716:/# uaserver 
Traceback (most recent call last):
  File "/usr/local/bin/uaserver", line 8, in <module>
    sys.exit(uaserver())
  File "/usr/local/lib/python3.8/site-packages/asyncua/tools.py", line 546, in uaserver
    sync.start_thread_loop()
AttributeError: module 'asyncua.sync' has no attribute 'start_thread_loop'
root@1bd3fc641716:/#
```

What am I missing?

driftregion, 2020-06-16:
This still a problem in the latest release 0.8.3, and seems unrelated to Docker.
It appears that `sync.start_thread_loop` and `sync.stop_thread_loop` have been removed.
I can get the server to run by removing calls to those functions in `tools.py`

swamper123, 2020-12-07:
Since there have been some changes since then, is it still an issue @driftregion @gkotian ?

gkotian, 2020-12-08:
Changing priorities have meant that I'm not working on OPC-related tasks at the moment. Closing this issue as obsolete would be fine by me. I will open a new one in the future if I run into trouble again.

closed 2020-12-09 by gkotian