FreeOpcUa/opcua-asyncio #107: Question: Set a object variable value from uamethod method?

linsjb, 2019-12-02:
I don't know it this is possible at all. If it is i'm not able to solve how to do it.

I want to create a uamethod on the server that i call from the client (no problem). In this method i want to set a value on a node variable (problem). Is this even possible? Has combed the internet for an answer without any success.

Short example with only the important bits below.The variable `variableToSet` is the var i want to set a value for **in** the uamethod `theMethod`.

I know i should probably do this with the parent. But exactly how?
Been printing out the parent to se what kind of info it hold's and in that way try to backward engineering it. But with no further success.

```python
@uamethod
def theMethod(parent, x):
await variableToSet.set_value(x)

async def main():
...
objects = server.get_objects_node()
variableToSet = await objects.add_variable(namespace, "theVarName", None)
```

Maybe this should be done with a subscription instead? 

Thanks in advance for any help and/or explanation. You guys it doing an incredible job with this.

mar-ar, 2019-12-02:
Just read the error. You have to create an async task for it, or just make the Method async def.

linsjb, 2019-12-02:
@mar-ar Thanks for your reply.
Yeah i know that. Have been testing it back and forth a few times. But the main problem, how i see it (or at least as i know) Is not the problem. The problem is to be able to access the `varToSet` inside the @uamethod.

I know that i get the node from the `parent` declaration in the @uamethod. But i can't get the idea how it works. 

Oh yeah.
the error message does not even state anything with the async/await at al. It rather states

```
NameError: name 'variableToSet' is not defined
```

Which is not that strange since the variable it self is not declared in the method.

oroulet, 2019-12-03:
somehting similar to this? but really learn some more pythnon and follow PEP8

```
class TOTO:
    def __init__(self, var):
         self.var = var

    @uamethod
    def theMethod(self, parent, x):
        await self.var.set_value(x) <- My problem

async def main():
    ...
    objects = server.get_objects_node()
    variableToSet = await objects.add_variable(namespace, "theVarName", None)
    toto = TOTO(variableToset)
    server.link_method(node_to_my_method, toto.theMethod
```

linsjb, 2019-12-04:
@oroulet Something like that should work for sure! Thanks.
Why didn't i think about a class. That's like an obvious solution.

Yeah. Kind of new in the Python game. Mostly been writing C++, Java, php and Js.

Thanks again!

closed 2019-12-04 by linsjb