FreeOpcUa/opcua-asyncio #111: Faults in the test script

mar-ar, 2019-12-16:
Hi guys,

I found a big problem with all the wreitten tests.
After each test we are deleting the subscriptions, but we leave the nodes and the eventgenerator at the server. Therefore we have some unexpected behavior. While a later test we got all nodes and everythin expect the subscriptions of the previous tests.

Here is one example:

    async def test_events_default(opc):
        evgen = await opc.server.get_event_generator()
        myhandler = MySubHandler()
        sub = await opc.opc.create_subscription(100, myhandler)
        handle = await sub.subscribe_events()
        tid = datetime.utcnow()
        msg = "this is my msg "
        evgen.trigger(tid, msg)
        ev = await myhandler.result()
        assert ev is not None  # we did not receive event
        assert ua.NodeId(ua.ObjectIds.BaseEventType) == ev.EventType
        assert 1 == ev.Severity
        assert (await opc.opc.get_server_node().get_browse_name()).Name == ev.SourceName
        assert opc.opc.get_server_node().nodeid == ev.SourceNode
        assert msg == ev.Message.Text
        assert tid == ev.Time
        await sub.unsubscribe(handle)
        await sub.delete()

After this test the server keeps to emit the BaseEvent which effects the later tests.

Correct code would be:

```
async def test_events_default(opc):
    evgen = await opc.server.get_event_generator()
    myhandler = MySubHandler()
    sub = await opc.opc.create_subscription(100, myhandler)
    handle = await sub.subscribe_events()
    tid = datetime.utcnow()
    msg = "this is my msg "
    evgen.trigger(tid, msg)
    ev = await myhandler.result()
    assert ev is not None  # we did not receive event
    assert ua.NodeId(ua.ObjectIds.BaseEventType) == ev.EventType
    assert 1 == ev.Severity
    assert (await opc.opc.get_server_node().get_browse_name()).Name == ev.SourceName
    assert opc.opc.get_server_node().nodeid == ev.SourceNode
    assert msg == ev.Message.Text
    assert tid == ev.Time
    await sub.unsubscribe(handle)
    await sub.delete()
    server_node = opc.server.get_server_node()
    await server_node.delete_reference(2041, 41)
```

I dont want to change all tests so I search a easy way to fit that. I wonder that there are so few problem with this behavior.

Maybe someone have a idea.

oroulet, 2019-12-16:
Well when I started writing tests I decided not to delete nodes (to save programming time) and not restart server (To save time running tests) so when we wrote tests we were always carefull to always create new nodes and only use our own nodes, not nods that may have been created by others.
But in you event case, you should either make the server restart, or make sure to stop the publishing of that particular event

mar-ar, 2019-12-16:
Yeah with the addition nodes I also see no problem, but with the events there is a problem.
I can only fix the effected tests when its ok for you. So we dont have to restart the server each time.

closed 2019-12-16 by mar-ar