FreeOpcUa/opcua-asyncio #119: Error on find_endpoint client call against a Milo OPC-UA server

aopenna, 2019-12-19:
When running client_mininal.py example against a Milo OPCUA server with security settings off ( security mode = None e security Policy = None) I got an error on find_endpoint call:

```
packages/asyncua/client/client.py", line 87, in find_endpoint
    raise ua.UaError("No matching endpoints: {0}, {1}".format(security_mode, policy_uri))
asyncua.ua.uaerrors._base.UaError: No matching endpoints: 1, http://opcfoundation.org/UA/SecurityPolicy#None
```

enviroment: MacOS 10.14.6
                     python version: Python 3.7.4

code :[client_minimal.py.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/3985820/client_minimal.py.zip)

```
import asyncio
import sys
sys.path.insert(0, "..")
import logging
from asyncua import Client, Node, ua

logging.basicConfig(level=logging.DEBUG)
_logger = logging.getLogger('asyncua')

async def main():
    url = 'opc.tcp://54.233.228.202:9000/wits-comgas/'
    async with Client(url=url) as client:
        await client.connect()

if __name__ == '__main__':
    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(main())
    loop.close()

```
below are the log (mode = DEBUG)

```
Alexandres-MacBook-Pro:~ apenna$ /Library/Frameworks/Python.framework/Versions/3.7/bin/python3 "/Users/apenna/Shared Folder Win10/comgas/client_minimal.py"

DEBUG:asyncio:Using selector: KqueueSelector
INFO:asyncua.client.client:connect
INFO:asyncua.client.ua_client.UaClient:opening connection
DEBUG:asyncio:connect <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('0.0.0.0', 0)> to ('54.233.228.202', 9000)
DEBUG:asyncio:poll took 29.096 ms: 1 events
DEBUG:asyncio:<socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('192.168.125.105', 63763), raddr=('54.233.228.202', 9000)> connected to 54.233.228.202:9000: (<_SelectorSocketTransport fd=7 read=polling write=<idle, bufsize=0>>, <asyncua.client.ua_client.UASocketProtocol object at 0x106340090>)
DEBUG:asyncio:poll 3999.081 ms took 27.307 ms: 1 events
INFO:asyncua.client.ua_client.UASocketProtocol:open_secure_channel
DEBUG:asyncua.client.ua_client.UASocketProtocol:Sending: OpenSecureChannelRequest(TypeId:FourByteNodeId(i=446), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeId(i=0), Timestamp:2019-12-19 21:56:11.521387, RequestHandle:1, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:OpenSecureChannelParameters(ClientProtocolVersion:0, RequestType:0, SecurityMode:1, ClientNonce:b'', RequestedLifetime:3600000))
DEBUG:asyncio:poll 3999.790 ms took 33.287 ms: 1 events
INFO:asyncua.client.ua_client.UaClient:create_session
DEBUG:asyncua.client.ua_client.UASocketProtocol:Sending: CreateSessionRequest(TypeId:FourByteNodeId(i=461), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeId(i=0), Timestamp:2019-12-19 21:56:11.561303, RequestHandle:2, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:10000, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:CreateSessionParameters(ClientDescription:ApplicationDescription(ApplicationUri:urn:freeopcua:client, ProductUri:urn:freeopcua.github.io:client, ApplicationName:LocalizedText(Encoding:2, Locale:None, Text:Pure Python Async. Client), ApplicationType:1, GatewayServerUri:None, DiscoveryProfileUri:None, DiscoveryUrls:[]), ServerUri:None, EndpointUrl:opc.tcp://54.233.228.202:9000/wits-comgas/, SessionName:Pure Python Async. Client Session1, ClientNonce:b'_(\xa4\x06\xa0i\xebu\xf2\xc3~c\xbc\xe1\xe3*"|\x07\xe4-\xb9\x0c\xafK\xddb\xa9\x82\x1c\xf1\x01', ClientCertificate:None, RequestedSessionTimeout:3600000, MaxResponseMessageSize:0))
DEBUG:asyncio:poll 9999.889 ms took 26.951 ms: 1 events
DEBUG:asyncua.client.ua_client.UaClient:CreateSessionResponse(TypeId:FourByteNodeId(i=464), ResponseHeader:ResponseHeader(Timestamp:2019-12-19 21:56:11.588000, RequestHandle:2, ServiceResult:StatusCode(Good), ServiceDiagnostics:DiagnosticInfo(Encoding:0, SymbolicId:None, NamespaceURI:None, Locale:None, LocalizedText:None, AdditionalInfo:None, InnerStatusCode:None, InnerDiagnosticInfo:None), StringTable:None, AdditionalHeader:None), Parameters:CreateSessionResult(SessionId:StringNodeId(ns=1;s=Session:c1e5d8eb-bd9f-4fb7-85b7-643eec79165a), AuthenticationToken:ByteStringNodeId(b=b'\xe6\x9c\x82yg\xd6VU\xd5/\x8f[\xf2]\xfb\xe6n^\xb6\x88\x1d!\xfd\xf5`\xa3\xbf\x0cl\xf4b\xad'), RevisedSessionTimeout:120000.0, ServerNonce:b' q\x16\xf9\xd3\x88bw\x8b\xcf9\xb3\x83\x81\xcf\xd7.\xb8\xa7\x85\\\xe3\x03\x8bal2\x8a\xf4\x1e\x81\xc9', ServerCertificate:b'0\x82\x04j0\x82\x03R\xa0\x03\x02\x01\x02\x02\x06\x01m\xda;U\x0e0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x0b\x05\x000v1$0"\x06\x03U\x04\x03\x0c\x1bEclipse Milo Example Server1\x150\x13\x06\x03U\x04\n\x0c\x0cdigitalpetri1\x0c0\n\x06\x03U\x04\x0b\x0c\x03dev1\x0f0\r\x06\x03U\x04\x07\x0c\x06Folsom1\x0b0\t\x06\x03U\x04\x08\x0c\x02CA1\x0b0\t\x06\x03U\x04\x06\x13\x02US0\x1e\x17\r191017000000Z\x17\r221017000000Z0v1$0"\x06\x03U\x04\x03\x0c\x1bEclipse Milo Example Server1\x150\x13\x06\x03U\x04\n\x0c\x0cdigitalpetri1\x0c0\n\x06\x03U\x04\x0b\x0c\x03dev1\x0f0\r\x06\x03U\x04\x07\x0c\x06Folsom1\x0b0\t\x06\x03U\x04\x08\x0c\x02CA1\x0b0\t\x06\x03U\x04\x06\x13\x02US0\x82\x01"0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x01\x05\x00\x03\x82\x01\x0f\x000\x82\x01\n\x02\x82\x01\x01\x00\x95\xc1\xeb\x8f4\xc6\xeeO\x12\x08\x85\xad\x11\x7f\x8a\xba\x1f\x81\xa9e\xe2\xa1\xa9b\xdb\x94`\xd5\xf0a\xa2\xc0C\xc2\xd2\xf8/L$m\xbc_\xc9\xe6\xf3\x0f\xd7\xc8\x7f\xe4\x91C\xe9\xf6r>\xb2\xbc\x01\x9f:i=4N+\xab\xee\xddo\xc3\x13>\xdc\xechef\xde\xfd\xd0\xf6\xd3\x8dF\xbej\xf7\xebWP@\xeef\x84\x80\x7fJ\xce\xd8"V\x8c\xd3\xa2\x1cq\xf4b\xef\x00n\xe8&\x88\xf4"\x1dMS\x94\x8c\xffZ\xc3n\xe2}\x05\x92I"\xa7\x88\x8b\x13~\x81J\x8cS\x89\xe6\x16\x0f\xeaZ\xbd\xd6\xec\xb5\x07\xe4\tA\x1d\xd3\x89\x0f\x04\xeb\xb4N\x82_\xd6T\x9a[D\x8aWi\x9d\xcd\x0c\xbf^\xf9\x80<\x08\x8aM\x1ex=dR\x11\xecIht}EX%\x9dAf\xd9p\x9f\xb4%\xa3\x04L\xb9\x07C\x07FI\xca\xb5\x0b\xcc\xee\x9b\x83\x1a\x164\x98\xb3U\x14\x91i\x80\xb9U{(\xc4\x14\xda\xea\xe7\xd8S\xf99\xfa\xe8\xe1\xff\x9f\xbb\xd9g\x8d\xac\xd7\x02\x03\x01\x00\x01\xa3\x81\xfd0\x81\xfa0\x1f\x06\x03U\x1d#\x04\x180\x16\x80\x14\x16\x8d\x13\xdcF\x1a5*N\xaf\xa7\x8d?\xe7\x1d\xe3_\x91u\xe50\x0c\x06\x03U\x1d\x13\x04\x050\x03\x01\x01\xff0\x0b\x06\x03U\x1d\x0f\x04\x04\x03\x02\x02\xfc0\x1d\x06\x03U\x1d%\x04\x160\x14\x06\x08+\x06\x01\x05\x05\x07\x03\x02\x06\x08+\x06\x01\x05\x05\x07\x03\x010~\x06\x03U\x1d\x11\x04w0u\x86Eurn:eclipse:milo:examples:server:be467060-14a8-46ab-9c2d-2cb37627c1eb\x82&root_adapter-opc-comgas_1.root_default\x87\x04\xac\x13\x00\x140\x1d\x06\x03U\x1d\x0e\x04\x16\x04\x14\x16\x8d\x13\xdcF\x1a5*N\xaf\xa7\x8d?\xe7\x1d\xe3_\x91u\xe50\r\x06\t*\x86H\x86\xf7\r\x01\x01\x0b\x05\x00\x03\x82\x01\x01\x00mJ\x1f\r\xf5U\x8aH\xef&{\xectn3\xd6%=\x85Uzg\x1d\xcf\xafShJ\xc2\x88\xc7\xec,\xa0\x1c\xa2\x8e\x7f\xe5\xcez\xee\x98jy\n\xfb..\x01\x1e\xfcN\x98\rd\xdaXh6\xb5\xa1<\xd0\xc2\x99<\xd4\xadZ7\xb0\xc8W\x97T\x1dRI\xb7\x89\xcc\x04\x9c\xf4\x13p\xf5c\xb5\xee\'y\xbe\t\x11\x7f\xc4\x12#\xaf9!~\xeb\xb7ji\x10\xach\x07\xc1\xa2rp^ZA\xa4h3\x91\x91L\xcf?X\xa5|d+\xcc\xd6[$lS\xb9S\x833\xc3\x03\xcbZ\xc8\xc7\xed\x04\xd6\x94\xf8#_\x16\x1f\x97\x87\xb4\x19\xe4e1\x9d\xc57\xfc\xa5\xcb\xc8b\xdb\x93\x8d\x87(\x80\xb7dD\x86\xe5@]*\xbe\xec\xc5\xb7g>-\xdb\x14\x0c\xb7A@?"\x0f~\x06\xdc\x91\xc4jI>LEi\xd2\x12Kq\x87+Sh\x0b\x07|\xce\x86a@\xc4\x00\x0f\x1f\xeb\x16wOW\xbfG\xf3\x9bK\x07^C\x90PU6\xf6\xd3c\xd9X\x94\xcf', ServerEndpoints:[], ServerSoftwareCertificates:[], ServerSignature:SignatureData(Algorithm:None, Signature:None), MaxRequestMessageSize:2097152))
INFO:asyncua.client.client:find_endpoint [] <MessageSecurityMode.None_: 1> 'http://opcfoundation.org/UA/SecurityPolicy#None'
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
Traceback (most recent call last):
  File "/Users/apenna/Shared Folder Win10/comgas/client_minimal.py", line 39, in <module>
    loop.run_until_complete(main())
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "/Users/apenna/Shared Folder Win10/comgas/client_minimal.py", line 14, in main
    async with Client(url=url) as client:
[... cut ...]
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/asyncua/client/client.py", line 342, in create_session
    ep = Client.find_endpoint(response.ServerEndpoints, self.security_policy.Mode, self.security_policy.URI)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/asyncua/client/client.py", line 87, in find_endpoint
    raise ua.UaError("No matching endpoints: {0}, {1}".format(security_mode, policy_uri))
asyncua.ua.uaerrors._base.UaError: No matching endpoints: 1, http://opcfoundation.org/UA/SecurityPolicy#None

```

aopenna, 2019-12-19:
trying do find more info I ran opcua-client and clicked on [connect options] -> [Query server capability] I got following results:

```
Alexandres-MacBook-Pro:comgas apenna$ opcua-client 
uaclient.uaclient - INFO - Endpoint 1:')
uaclient.uaclient - INFO -   Endpoint URL: opc.tcp://localhost:9000/wits-comgas')
uaclient.uaclient - INFO -   Application URI: urn:eclipse:milo:examples:server:be467060-14a8-46ab-9c2d-2cb37627c1eb')
uaclient.uaclient - INFO -   Product URI: urn:eclipse:milo:example-server')
uaclient.uaclient - INFO -   Application Name: Adapter OPC UA Server COMGAS')
uaclient.uaclient - INFO -   Application Type: ApplicationType.Server')
uaclient.uaclient - INFO -   Discovery URL: opc.tcp://localhost:9000/wits-comgas/discovery')
uaclient.uaclient - INFO -   Server Certificate: commonName=Eclipse Milo Example Server, organizationName=digitalpetri, organizationalUnitName=dev, localityName=Folsom, stateOrProvinceName=CA, countryName=US (self-signed), 2019-10-17 00:00:00 - 2022-10-17 00:00:00')
uaclient.uaclient - INFO -   Security Mode: MessageSecurityMode.None_')
uaclient.uaclient - INFO -   Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None')
uaclient.uaclient - INFO -   User policy: anonymous')
uaclient.uaclient - INFO -     Token type: UserTokenType.Anonymous')
uaclient.uaclient - INFO -   User policy: username')
uaclient.uaclient - INFO -     Token type: UserTokenType.UserName')
uaclient.uaclient - INFO -     Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic256')
uaclient.uaclient - INFO -   User policy: certificate')
uaclient.uaclient - INFO -     Token type: UserTokenType.Certificate')
uaclient.uaclient - INFO -     Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic256')
uaclient.uaclient - INFO -   Transport Profile URI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary')
uaclient.uaclient - INFO -   Security Level: 32')
uaclient.uaclient - INFO - ')
uaclient.uaclient - INFO - Endpoint 2:')
uaclient.uaclient - INFO -   Endpoint URL: opc.tcp://localhost:9000/wits-comgas')
uaclient.uaclient - INFO -   Application URI: urn:eclipse:milo:examples:server:be467060-14a8-46ab-9c2d-2cb37627c1eb')
uaclient.uaclient - INFO -   Product URI: urn:eclipse:milo:example-server')
uaclient.uaclient - INFO -   Application Name: Adapter OPC UA Server COMGAS')
uaclient.uaclient - INFO -   Application Type: ApplicationType.Server')
uaclient.uaclient - INFO -   Discovery URL: opc.tcp://localhost:9000/wits-comgas/discovery')
uaclient.uaclient - INFO -   Server Certificate: commonName=Eclipse Milo Example Server, organizationName=digitalpetri, organizationalUnitName=dev, localityName=Folsom, stateOrProvinceName=CA, countryName=US (self-signed), 2019-10-17 00:00:00 - 2022-10-17 00:00:00')
uaclient.uaclient - INFO -   Security Mode: MessageSecurityMode.SignAndEncrypt')
uaclient.uaclient - INFO -   Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256')
uaclient.uaclient - INFO -   User policy: anonymous')
uaclient.uaclient - INFO -     Token type: UserTokenType.Anonymous')
uaclient.uaclient - INFO -   User policy: username')
uaclient.uaclient - INFO -     Token type: UserTokenType.UserName')
uaclient.uaclient - INFO -     Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic256')
uaclient.uaclient - INFO -   User policy: certificate')
uaclient.uaclient - INFO -     Token type: UserTokenType.Certificate')
uaclient.uaclient - INFO -     Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic256')
uaclient.uaclient - INFO -   Transport Profile URI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary')
uaclient.uaclient - INFO -   Security Level: 136')
uaclient.uaclient - INFO - ')
```

themightyoarfish, 2023-04-21:
Any solution to this?

AndreasHeine, 2023-04-22:
isnt it sessionless request? i will look into that soon!