FreeOpcUa/opcua-asyncio #123: AttributeError: 'Node' object has no attribute 'read_value'

Elthas17, 2020-01-06:
Dear,

Thanks for opcua-asyncio

In your examples you comment out a line such as:
https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-example.py

Line 46 #var.read_value() # get value of node as a python builtin 

When connected to an OPC UA server (Beckhoff CX8091) this fuction yields:

> print(var.read_value()) # get value of node as a python builtin
> AttributeError: 'Node' object has no attribute 'read_value'
> INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection

 var.get_value() yields the expected result instead

My code snippit
```
var = client.get_node(ua.NodeId('MAIN.mynode', 4)) 
print(var.read_value()) # gives an error
print(await var.get_value()) # works fine
```

oroulet, 2020-01-06:
Get_value just got renamed to read_value. Run asyncua from master if you want read_value

Elthas17, 2020-01-07:
I installed the latest version with pip, so pip does not hold the latest version of the master branch?

oroulet, 2020-01-07:
No. Pip has released versions.

tukotech, 2020-08-07:
Just installed 0.8.4 using pip, <node>.read_value() works. Should this issue be closed?

swamper123, 2020-12-16:
Since there are no complains since months and the renaming was clarified, the Case seems to be closeable 🕵️

closed 2020-12-16 by swamper123