FreeOpcUa/opcua-asyncio #129: Ex/Import Server via XML - dealing with Methods

swamper123, 2020-01-13:
Hello together,

I try to export a Server via `await server.export_xml_by_ns(path,[namespaces])` and import this xml file in another program via `await server.import_xml(path)`.

It works fine with values, folders etc. (haven't tried it with events tbh), but when it comes to methods it raises this error:
`asyncua.ua.uaerrors._auto.BadParentNodeIdInvalid: The parent node id does not to refer to a valid node.(BadParentNodeIdInvalid)`

I know that the method should be linked to a python function again, but this doesn't seem to be the problem, does it? Am  I missing something?

swamper123, 2020-01-14:
It seems that the exported Methods haven't got any References at all. Could this be caused by the xmlexporter?

oroulet, 2020-01-14:
Might be but here. I do not like the xml parsing code any PR to improve things is welcome

swamper123, 2020-01-14:
I just sorted my nodes in Methods and non-methods and exported the non-methods part in an XML-file. In this file appeared the correct UaVariables for Output- and InputArguments referencing to the MethodNode.
After Adding the upper XML-Block into it and imported again, the server seems to be fine and even in the original state like before.

swamper123, 2020-01-15:
Found a wrong indent in  `async def add_etree_method(self, node):`
The line `await self._add_ref_els(obj_el, node)` needs to be outdented. I will make a request later.

oroulet, 2020-01-15:
Great!

closed 2020-01-15 by swamper123