FreeOpcUa/opcua-asyncio #133: Get Tree of objects

luisfmelo, 2020-01-22:
Hi,

first of all, thanks for the library. It seems really good in my first days studying it. 

I have one question. Is it possible to retrieve the tree of objects of a server?

oroulet, 2020-01-22:
You never went to get the entire tree it is too big. But using get_children () you can easily navigate tree

oroulet, 2020-01-22:
You can also export nodes to xml

swamper123, 2020-01-23:
Maybe you can grab the nodes of a namespace you want via get_nodes_of_namespace, maybe filter it and then use the exporter.built_etree() method to get a tree.
Or export them in an xml file and try to find your ways through the structure.

swamper123, 2020-12-15:
Seems to be solved by inactivity. Case closed 🕵️

closed 2020-12-15 by swamper123