FreeOpcUa/opcua-asyncio #134: self.value_callback - for AttributeValue requests? And how to add them?

swamper123, 2020-01-23:
Ahoi,

just wanted to ask if `self.value_callback` shall be a possibility to add (lambda) functions or coroutines as callback, in case of an AttributeValue request? If yes, is there a clean way to add them in anywhere?

```
class AttributeValue(object):

    def __init__(self, value):
        self.value = value
        self.value_callback = None
        self.datachange_callbacks = {}

    def __str__(self):
        return f"AttributeValue({self.value})"

    __repr__ = __str__

```

swamper123, 2020-01-29:
So I took some stuff from here and pulled out the dict and datachange_callback stuff:
```
    def add_datachange_callback(self, nodeid, attr, callback):
        self.logger.debug("set attr callback: %s %s %s", nodeid, attr, callback)
        if nodeid not in self._nodes:
            return ua.StatusCode(ua.StatusCodes.BadNodeIdUnknown), 0
        node = self._nodes[nodeid]
        if attr not in node.attributes:
            return ua.StatusCode(ua.StatusCodes.BadAttributeIdInvalid), 0
        attval = node.attributes[attr]
        self._datachange_callback_counter += 1
        handle = self._datachange_callback_counter
        attval.datachange_callbacks[handle] = callback
        self._handle_to_attribute_map[handle] = (nodeid, attr)
        return ua.StatusCode(), handle
```

and made my own method, to make adding callbacks in Attributes possible:

```
    def add_value_callback_to_node(self, nodeid, attr, callback):
        if nodeid not in self.iserver.aspace._nodes:
            return ua.StatusCode(ua.StatusCodes.BadNodeIdUnknown), 0
        node = self.iserver.aspace._nodes[nodeid]
        if attr not in node.attributes:
            return ua.StatusCode(ua.StatusCodes.BadAttributeIdInvalid), 0
        attval = self.iserver.aspace._nodes[nodeid].attributes[attr]
        attval.value_callback = callback
        return ua.StatusCode()
```
I putted this stuff in my Server Class. Maybe it need some changes and could be used as Adressspace Method as well?!

swamper123, 2020-01-29:
I found out, that in internal session,  `attribute_service` is used as coroutines, but is executing sync methods. 
So I will try to update `get_attribute_value` and `set_attribute_value` to be async, so even Coroutines could be added `value_callback`. This will need changes in `AttributeService` read and write and needs an await in `internal_session` for read and write.

oroulet, 2020-01-29:
I do not follow, but send a pr

swamper123, 2020-01-31:
#138

closed 2020-12-15 by swamper123