FreeOpcUa/opcua-asyncio #135: Server update variable

brandonfwl, 2020-01-27:
What would be the appropriate way to update a variable on an interval using the server?  I have a few weather api calls that I'd like to update on the server.

`w_wind_speed = await myobj.add_variable(idx, 'Wind Speed', wind_speed)`

I'd like these values to be updated by the server every x seconds for consumption.  Is there a "best practice" way to do this?

Any help would be great.

oroulet, 2020-01-27:
Isn't it what the server-minimal.py example do? And server-example.py

swamper123, 2020-01-28:
You can do it in a loop like in the examples.
Another way I can imagine is to use `asyncio.ensure_future(coro,*,loop=None)` method, write your own coroutine and add an` asyncio.call_later(delay, callback, *args)` at the end of the coroutine. It will run, until somebody is cancelling it or something crashes (hopefully not).

oroulet, 2020-01-28:
@swamper123 asyncio.create_task() in newer pyuthon versions and make sure to await the task at end of program....

brandonfwl, 2020-01-29:
Thanks a ton guys for the help.  One other question although perhaps it's more the consumer that's the issue.  When I define a variable as writable by the client and then write to said variable on the python server side it appears to accept the change but the consumer never registers the updated value.  Has anyone seen this?  I'm testing with Tridium Niagara building automation software.  I'll test with the opc-client gui when possible.

Again thank for the guidance.

swamper123, 2020-01-29:
@brandonfwl I think there is no auto-update of value, when you write a new value into a variable. You may need another read-request to get the value. I think there is a way to get something like a data_changed notification from a node, but I don't know the used event/method at the moment. If I find something, I will let you know.

oroulet, 2020-01-29:
Writing the value of a node generate a data change event for sure. This what is done in many tests and in example-server.py

Bawwa-cmd, 2021-03-09:
> @brandonfwl I think there is no auto-update of value, when you write a new value into a variable. You may need another read-request to get the value. I think there is a way to get something like a [...]

Hey there, Did you find a way for this? @swamper123 @brandonfwl

AndreasHeine, 2021-03-09:
Subscription!
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.13.1/

swamper123, 2021-03-10:
@Bawwa-cmd https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-subscription.py

Bawwa-cmd, 2021-03-10:
> @Bawwa-cmd https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-subscription.py

Great :)Thank you!

Bawwa-cmd, 2021-03-10:
> 
> 
> @Bawwa-cmd https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-subscription.py

I am getting an error when writing to the node as;

  File "C:\Users\lokadmin\anaconda3\lib\site-packages\asyncua\ua\uatypes.py", line 220, in check
    raise UaStatusCodeError(self.value)

**BadOutOfRange: "The value was out of range."(BadOutOfRange)**

Does anybody know why?

swamper123, 2021-03-10:
I ran server-minimal.py and client-subscription.py. No error occurred. 
Please share the code you used, what you expected, etc. (server & client), like mentioned in the Bug issue Template.
Maybe even make another issue, since this issue seems to be solved.

Bawwa-cmd, 2021-03-10:
> Have you checked if the value is even legit for the Nodes you are writing to (check NodeType)?

Yes, I did and it is legit. When I use set_value(3.869906878e-001), It works right. But when I assign the variable which includes this value, It gives the above error.

swamper123, 2021-03-10:
> Yes, I did and it is legit. When I use set_value(3.869906878e-001), It works right. But when I assign the variable which includes this value, It gives the above error.

In general I would recommened:
- either use ua.Variant(YOUR_VALUE, ua.VariantType.YOUR_EXPECTED_TYPE) as value for write_value() (same as get_value(), just the newer name) or
- use the varianttype parameter of write_value(YOUR_VALUE, varianttype=ua.VariantType.YOUR_EXPECTED_TYPE)
 
Then you are sure, that the value is interpreted correctly. Check the expected type on server side.

Bawwa-cmd, 2021-03-11:
> * either use ua.Variant(YOUR_VALUE, ua.VariantType.YOUR_EXPECTED_TYPE) as value for write_value() (same as get_value(), just the newer name) or
> * use the varianttype parameter of [...]

- This is really helpful. Thanks to you I found my error. It was an error with the data type of value I gave to the node. Now it is working right. Thanks again :)

swamper123, 2021-03-11:
Guess both cases are solved now and can be closed 🕵️

closed 2021-03-11 by swamper123