FreeOpcUa/opcua-asyncio #139: Subscibing to change events forever

kunal-arora, 2020-02-05:
I am using the code in the [client-subscription.py](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-subscription.py). I want to watch change values in a forever loop.

Is that a bad design pattern ? If yes then how should I watch event changes for infinite time.

swamper123, 2020-02-05:
So you made  `await subscription.subscribe_data_change(nodes)` and then an infinity `await.sleep(0)` loop?

kunal-arora, 2020-02-05:
For now I am just using the code in the file mentioned above, but my main objective is to keep subscription looking for change values forever.

swamper123, 2020-02-05:
In that case you have to add `await.sleep(x)` (see line 44) in an infinity loop (or even nicer would be with a break condition by something else, to finish clearly). `x` would be any number greater equals 0.
So you would wait, until something crashes or you finish with a break condition, unsub your subscription and  close the connection.

swamper123, 2020-02-05:
In that case you have to add an infinity loop (or even nicer would be with a break condition by something else, to finish clearly) around `await asyncio.sleep(10)`  (see line 44).
So you would wait, until something crashes (unclean way) or you finish with a break condition, unsub your subscription and close the connection, which would be the clean way.

kunal-arora, 2020-02-05:
I did something like 
```
if(True):
       await asyncio.sleep(10)
```

I get an error after 10 seconds which looks like
```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\wamp64\www\opcua\PlantDataProcessor\venv\lib\site-packages\asyncua\client\ua_client.py", line 79, in _process_received_data
    self._process_received_message(msg)
  File "C:\wamp64\www\opcua\PlantDataProcessor\venv\lib\site-packages\asyncua\client\ua_client.py", line 93, in _process_received_message
    self._call_callback(msg.request_id(), msg.body())
  File "C:\wamp64\www\opcua\PlantDataProcessor\venv\lib\site-packages\asyncua\client\ua_client.py", line 163, in _call_callback
    raise ua.UaError(f"Future for request id {request_id} is already done")
asyncua.ua.uaerrors._base.UaError: Future for request id 33 is already done
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
INFO:asyncio:poll 9985.000 ms took 9985.000 ms: timeout
WARNING:asyncua.client.ua_client.UaClient:disconnect_socket was called but connection is closed
ERROR:asyncua:error
Traceback (most recent call last):
  File "C:\wamp64\www\opcua\PlantDataProcessor\scripts\opc_connection.py", line 106, in initialiseOPC
    await asyncio.sleep(10)
  File "C:\wamp64\www\opcua\PlantDataProcessor\venv\lib\site-packages\asyncua\client\client.py", line 75, in __aexit__
    await self.disconnect()
  File "C:\wamp64\www\opcua\PlantDataProcessor\venv\lib\site-packages\asyncua\client\client.py", line 228, in disconnect
    await self.close_session()
  File "C:\wamp64\www\opcua\PlantDataProcessor\venv\lib\site-packages\asyncua\client\client.py", line 471, in close_session
    return await self.uaclient.close_session(True)
  File "C:\wamp64\www\opcua\PlantDataProcessor\venv\lib\site-packages\asyncua\client\ua_client.py", line 312, in close_session
    data = await self.protocol.send_request(request)
  File "C:\wamp64\www\opcua\PlantDataProcessor\venv\lib\site-packages\asyncua\client\ua_client.py", line 140, in send_request
    timeout if timeout else None
  File "C:\Users\USER\AppData\Local\Programs\Python\Python37\Lib\asyncio\tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

kunal-arora, 2020-02-05:
I followed what you said and just using a while(True) works fine, but is it a reliable thing to do ?

swamper123, 2020-02-05:
Just a while(True): is not the nicest way, because you can just break the loop by Exceptions. Add a try/except block, so your program will finish a clean way. If you want to subscribe for datachanges forever, this would be the way.
I don't know at the moment, if the server has something like a "subscription timeout", but I don't think so.

kunal-arora, 2020-02-05:
okay thanks for the quick help.
Cheers!

closed 2020-02-11 by chrisjbremner