FreeOpcUa/opcua-asyncio #141: Client timeout on node.call_method()

jsbruglie, 2020-02-06:
Would it be possible to expose the `timeout` parameter in the function `UASocketProtocol.send_request` to higher-level functions, such as `UaClient.call()`?
Currently, if I call `Node.call_method()` the timeout will always be 10 seconds.

In my work we have to communicate with a single-threaded simulator with built-in OPC-UA server, which causes the server to block during complex simulation steps, so I needed to increase the timeout.

Should `timeout` be available from `Node.call_method()` for instance, or is there a better way of achieving a configurable timeout for these operations?

swamper123, 2020-02-14:
I  think the Clients have got a timeout parameter. You may checkout `asyncio.wait_for()` [in the docs](https://docs.python.org/3/library/asyncio-task.html#timeouts)

jsbruglie, 2020-02-14:
Thank you for your answer.

In this case, I don't think the client `timeout` parameter is even used.
The problem is that whenever you execute `Node.call_method` from a client you end up calling the method `UASocketProtocol.send_request` with default arguments:
`async def send_request(self, request, timeout=10, message_type=ua.MessageType.SecureMessage)`, in `ua_client.py`.
So, the explicit call to `asyncio.wait_for` that is in this function seems to be always called with a timeout of 10 seconds, regardless of how you configure the client.

Am I missing something?

oroulet, 2020-02-14:
@jsbruglie should be possible. you need to expose the extra argument timeout in call methods of ua_client and internall_session. Ignore the parameter in InternalSession but use it in UaClient to call send_request
you also need to add and pass that argument in _call_method and call_method_full and call method. Not sure it is worth to go up to Node.call_method

But generally long network call are a very bad idea... so many things can happen in the meantime...

jsbruglie, 2020-02-14:
Thank you @oroulet , that's what I've done for the moment.

I see your point and have to agree that allowing for large timeouts is not a desirable feature for everyone.
Unfortunately I really can't go around it, as the server I have to work with can essentially block for a long time while it performs computations, and I really have no way of changing it... 

So, I'd say I'll just have to use a modified version of the source for my particular use case.
Thank you for your input and help, and great job with this package: it's become the key element in my attempt to automate some PLC testing at CERN

jsbruglie, 2020-02-14:
I'll close the issue unless there are more suggestions, or further discussion.

oroulet, 2020-02-14:
anyway great if you can make a PR, might be usefull

jsbruglie, 2020-02-14:
I can also do a PR if you want, which value should I use as the timeout for these requests? Should it be `UaClient._timeout` ?

oroulet, 2020-02-15:
you should add a new parameter to call_method which is called timeout or method_timeout

jsbruglie, 2020-02-19:
@oroulet  It could be that I misunderstood the structure of the module, but it seems that `call_method` can be used both from client and server.
From the client you eventually get to `UASocketProtocol.send_request` which has a timeout.
However, from the server there seems to be no operation that can timeout.
So I would effectively be exposing an argument for a feature that's not implemented.
I can still do the PR in any case.

jsbruglie, 2020-09-18:
Closing this because at the time I saw no clear way on how to achieve this for clients, as the affected bit of code is shared for servers. In any case, this whole problem was caused by a single-threaded program with built-in OPC-UA server that would hang when performing internal computations. We eventually solved this with a work-around directly on the proprietary server.

closed 2020-09-18 by jsbruglie