FreeOpcUa/opcua-asyncio #146: Exception when trying to disconnect from the server with sync client.

TilmanK, 2020-02-17:
The disconnect method of the sync Client is obvioulsy broken. When calling disconnect, I'm getting the following exception:

`2020-02-17 10:00:17.626  Thread-10  -                    client - INFO     - disconnect
2020-02-17 10:00:17.626  Thread-10  -                 ua_client - INFO     - Request to close socket received
2020-02-17 10:00:17.626  Thread-10  -                 ua_client - INFO     - Socket has closed connection
Traceback (most recent call last):
  File "C:\Users\krumm-ti\PycharmProjects\LabMonitor\views\mainwindow.py", line 187, in closeEvent
    OpcClient.inst().disconnect()
  File "C:\Users\krumm-ti\PycharmProjects\LabMonitor\opc_client.py", line 170, in disconnect
    self._client.disconnect()
  File "C:\Users\krumm-ti\PycharmProjects\LabMonitor\venv\lib\site-packages\asyncua\sync.py", line 116, in disconnect
    self.tloop.post(self.aio_obj.disconnect())
  File "C:\Users\krumm-ti\PycharmProjects\LabMonitor\venv\lib\site-packages\asyncua\sync.py", line 49, in post
    return futur.result()
  File "C:\Users\krumm-ti\AppData\Local\Programs\Python\Python37-32\lib\concurrent\futures\_base.py", line 423, in result
    raise CancelledError()
concurrent.futures._base.CancelledError`

Minimal example:

`from asyncua.sync import Client

client = Client( "your uri here")
client.connect()
client.disconnect()
`

Can you please check what causes this? I wasn't able to figure it out by myself.

oroulet, 2020-02-17:
In fact it looks ok. To stop tasks a cancel exception is raised. But maybe we should catch it so users do not get confused

TilmanK, 2020-02-17:
Ah ok. Yes, indeed we should to micmic the behaviour of the old sync client. Catching the exception works as expected and the client disconnects.

I'll submit a PR tonight if you want.

oroulet, 2020-02-17:
My problem is that I am not really sure where it should be catched. The doc sais one should not catch CancelledError. This is true since it might make tasks not killable. But maybe inside disconnect is fine...

TilmanK, 2020-02-17:
Yes, exactly in the disconnect method of the sync Client. I'd expect the method to behave like the one from opcua sync lib and any difference will make migrations more complicated. 

And the exception doesn't make any sense when using sync IMHO.

TilmanK, 2020-02-17:
Another question is, if we should raise an exception when we do not catch the CancelledError on calling disconnect? If this is what we expect, then this means that no exception is a problem, right?

oroulet, 2020-02-17:
agree that exception in syn api does not make sense anyway, soyou can catch there. 
Concerning the async version, I am unsure, but I suppose we should catch there and do not raise anything in case of CanceledError

closed 2020-02-19 by TilmanK