FreeOpcUa/opcua-asyncio #149: uahistory read does not work

luisfmelo, 2020-02-20:
I tried:
`uahistoryread -u "opc.tcp://opcuaserver.com:48010" -n "ns=2;s=Demo.History.ByteWithHistory" --starttime "2020-02-20 11:04" --endtime "2020-02-20 11:06"`

The result:
```
Reading raw history of node Node(StringNodeId(ns=2;s=Demo.History.ByteWithHistory)) at opc.tcp://opcuaserver.com:48010; start at 2020-02-20 11:04:00, end at 2020-02-20 11:06:00

Source timestamp               Status     Value
[...]Traceback[...]
TypeError: 'coroutine' object is not iterable
sys:1: RuntimeWarning: coroutine 'Node.read_raw_history' was never awaited
```

I will try to fix it and make a PR.

oroulet, 2020-02-20:
Thanks. Also if it does not work then there is a test missing...

luisfmelo, 2020-02-20:
Not sure. I did not find any tests for the tools. Is there any test?

oroulet, 2020-02-20:
All tests under tests folder. Should be one called history.... And obviously there is something missing there

PR #150 (2020-02-20): Fixed uahiostoryread
Fixed #149 

I did not find any tests to fix.
closed 2020-02-20 by oroulet