FreeOpcUa/opcua-asyncio #160: Load_type_definitions exceptions

Naisunev, 2020-03-11:
When doing a load_type_definitions() request to a Siemens PLC OPC-UA server configured by the TIA portal, the XML is badly formatted since it contains a <xml /> tag before the rest of the XML. A simple search/replace to remove this part could prevent the error. 

Also when loading the type definitions the structs are somehow not classified as arrays even though they are in the supplied XML structure. I haven't traced through the client yet to see why/where this bug is introduced. I'll get back to you.

With another OPC-UA server I get the exception: BadResponseTooLarge. when downloading the XML structure. The default setting in the client is: "params.MaxResponseMessageSize = 0  # means no max size". When viewing the node in UA-Expert i see this reported length: len=175766

oroulet, 2020-03-11:
that is many issues in one report. Yes there might be bug and Yes any help is welcome ;-)

cailloumajor, 2020-05-02:
Hello, I also encounter problems in a quite similar context. First of all, let's describe it as best as I can : I am developing a bridge from OPC-UA to WebSocket for an homemade SCADA. The OPC-UA part comprises the following elements :
* Siemens S7-1500 OPC-UA server (actually simulated with S7-PLCSIM Advanced)
* CPython 3.8.2
* asyncua 0.8.2
You can find the provisional code I wrote for OPC-UA handling there : https://gist.github.com/cailloumajor/38e06836b126040f8ecbf45b1d8faccf

The 2 problems I encountered were the following (I got the second problem after fixing the first) :
1. Exception `ValueError: Unicode strings with encoding declaration are not supported. Please use bytes input or XML fragments without declaration.` raised from:
https://github.com/FreeOpcUa/opcua-asyncio/blob/870dc85b8e506b1ca465435973e1283f0d120bcc/asyncua/common/structures.py#L138
2. Following exception (end of the traceback):
```python
  File "/workspaces/opcua-graphql-bridge/.venv/lib/python3.8/site-packages/asyncua/common/structures.py", line 327, in _generate_python_class
    exec(code, env)
  File "<string>", line 3
    class "udtLineSupervision"."machineState"(object):
          ^
SyntaxError: invalid syntax
```

I am going to create one PR with some more explanations for each of the above problems. They correspond exactly with the modifications I made to fix those problems.

cailloumajor, 2020-05-02:
After applying above changes, the data I get from subscription is structured exactly as in the PLC 🎉

closed 2023-03-29 by schroeder-