FreeOpcUa/opcua-asyncio #163: Why does send_request in UaClient.publish have timeout=0?

chrisjbremner, 2020-03-12:
I am trying to troubleshoot the frequent [`Future for request id {request_id} is already done: {body}`](https://github.com/FreeOpcUa/opcua-asyncio/blob/5cadb8194ccd8a35622608a590f1c651351f5f75/asyncua/client/ua_client.py#L162-L163) errors I am seeing when using the sync client. The errors seem to occur only on `PublishResponse` messages, and my understanding of [this logic](https://github.com/FreeOpcUa/opcua-asyncio/blob/5cadb8194ccd8a35622608a590f1c651351f5f75/asyncua/client/ua_client.py#L475) is that the `PublishRequest` gets created and sent, but then won't [this](https://github.com/FreeOpcUa/opcua-asyncio/blob/5cadb8194ccd8a35622608a590f1c651351f5f75/asyncua/client/ua_client.py#L138-L141) always timeout and raise a `TimeoutError` because the timeout is 0? This then cancels the future with a `CancelledError` which I suspect then causes this error to occur whenever a `PublishResponse` message arrives. I'm curious whether this timeout makes sense to be 0, or if this is a holdover from `python-opcua` and this timeout argument can be removed for publishing? @cbergmiller I tagged you since you show up in the blame for this one, but it looks like it might be the change might originate [here](https://github.com/FreeOpcUa/python-opcua/commit/9e5a26a97f1079a5db1f55b365411c5e2b5d0317). Thanks for the help!

oroulet, 2020-03-12:
Great you are looking at it. I have never had that error but I see some people report it. 0 as timeout means that there is no timeout. This is the safest to do. Although there id somewhere a max time between publish request set so we could reuse that value. I cannot see how it can be related to that error

oroulet, 2020-03-12:
Can it be that you only see that request when closing? It might be that the future has already been cancelled when we receive a last response from server just before closing

chrisjbremner, 2020-03-12:
I think if we want no timeout then we want to use `None` rather than 0. It appears to timeout immediately when using 0:

```python
await asyncio.wait_for(asyncio.get_event_loop().create_future(), timeout=0) # raises TimeoutError
await asyncio.wait_for(asyncio.get_event_loop().create_future(), timeout=None)  # hangs indefinitely
```

chrisjbremner, 2020-03-12:
> Can it be that you only see that request when closing?

No, I see this during normal operation as well

chrisjbremner, 2020-03-16:
Ah I see, I didn't fully appreciate that the line said `timeout if timeout else None`, so it should wait indefinitely if `timeout=0`. Any other ideas of what to look for to determine why my tasks are getting cancelled? If not, I can go ahead and close and open something new when I have another lead.

closed 2020-03-17 by chrisjbremner