FreeOpcUa/opcua-asyncio #164: Bad_TcpMessageTooLarge error when using release `0.8.2`

flyingvince, 2020-03-14:
Getting an error message `Received an error: MessageAbort(error:StatusCode(BadTcpMessageTooLarge), reason:Bad_TcpMessageTooLarge (code=0x80800000, description="Chunk size (132) exceeded maximum (-1)"))` when trying to connect to a OPC UA server.

## Current Behavior
The full log and traceback is as follows:

Received an error: MessageAbort(error:StatusCode(BadTcpMessageTooLarge), reason:Bad_TcpMessageTooLarge (code=0x80800000, description="Chunk size (132) exceeded maximum (-1)"))
Received an error: MessageAbort(error:StatusCode(BadTcpMessageTooLarge), reason:Bad_TcpMessageTooLarge (code=0x80800000, description="Chunk size (132) exceeded maximum (-1)"))
Exception raised while parsing message from server
Traceback (most recent call last):
  File "/var/lang/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 158, in _call_callback
    self._callbackmap[request_id].set_result(body)
KeyError: 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/lang/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 79, in _process_received_data
    self._process_received_message(msg)
  File "/var/lang/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 98, in _process_received_message
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/var/lang/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 161, in _call_callback
    f"No request found for request id: {request_id}, pending are {self._callbackmap.keys()}"
asyncua.ua.uaerrors._base.UaError: No request found for request id: 0, pending are dict_keys([1])
disconnect_socket was called but connection is closed
Traceback (most recent call last):
  File "opc_ua_server_test.py", line 44, in <module>
    loop.run_until_complete(query_opc_ua())
  File "/var/lang/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "opc_ua_server_test.py", line 31, in query_opc_ua
    async with client_opc as client:
  File "/var/lang/lib/python3.7/site-packages/asyncua/client/client.py", line 71, in __aenter__
    await self.connect()
  File "/var/lang/lib/python3.7/site-packages/asyncua/client/client.py", line 217, in connect
    await self.open_secure_channel()
  File "/var/lang/lib/python3.7/site-packages/asyncua/client/client.py", line 267, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "/var/lang/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 271, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "/var/lang/lib/python3.7/site-packages/asyncua/client/ua_client.py", line 202, in open_secure_channel
    self.timeout
  File "/var/lang/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError

## Possible Solution
The temporary fix is to revert back to version `0.8.1`, which is able to connect to the OPC UA server and read data successfully.

## Steps to Reproduce
1. Use Docker related files available [here](https://www.dropbox.com/sh/i97e5blempr6i4x/AACMCM5ATnrr31PoTUZmPZ8ka?dl=0)

2. Build a Docker image using the Dockerfile as is: `docker build . -t async-ua:requirements-ok`
3. In the Dockerfile, comment out line 15 and uncomment line 16 (=use `requirements-fails.txt`) for the Docker build
4. Build a Docker image using the revised Dockerfile: `docker build . -t async-ua:requirements-fails`
5. Test by running `docker run -it async-ua:requirements-ok` and later run `docker run -it async-ua:requirements-fails`
6. The first Docker container is able to connect to the OPC UA server and the logs are:

2020-03-14 00:44:51,926 - INFO - sys.version = 3.7.3 (default, May 31 2019, 14:47:02)
[GCC 4.8.3 20140911 (Red Hat 4.8.3-9)]
2020-03-14 00:44:51,998 - INFO - Node(TwoByteNodeId(i=84))
2020-03-14 00:44:52,014 - INFO - The triangle value = -1.3333333055583805
2020-03-14 00:44:53,031 - INFO - The triangle value = -1.5999999796457232
2020-03-14 00:44:54,046 - INFO - The triangle value = -1.866666653733066
2020-03-14 00:44:55,061 - INFO - The triangle value = -1.8666666721795913
2020-03-14 00:44:56,075 - INFO - The triangle value = -1.5999999980922488
2020-03-14 00:44:57,072 - INFO - The triangle value = -1.3333333998958972
2020-03-14 00:44:58,086 - INFO - The triangle value = -1.066666725808554
2020-03-14 00:44:59,101 - INFO - The triangle value = -0.8000000517212114
2020-03-14 00:45:00,117 - INFO - The triangle value = -0.5333333776338687
2020-03-14 00:45:01,132 - INFO - The triangle value = -0.266666703546526

7. The second Docker container return the error log and traceback detailed earlier

## Context (Environment)
The OPC UA server being used is the [Prosys OPC UA Simulation Server](https://www.prosysopc.com/products/opc-ua-simulation-server/)

oroulet, 2020-05-16:
the onlu possible impacting change I can find i at that line:
   def __init__(self):
        self.ProtocolVersion = 0
        self.ReceiveBufferSize = 2**32 - 1
        self.SendBufferSize = 2**32 - 1
        self.MaxMessageSize = 0  # No limits
        self.MaxChunkCount = 0  # No limits

but none of the values here are set to -1.... you can try to play with them anyway

swamper123, 2020-12-16:
There have been a bunch full of changes since them.
@flyingvince Is this issue still active? Otherwise I would like to close it.

flyingvince, 2020-12-17:
Hi @swamper123, as per your suggestion please close the issue.

swamper123, 2020-12-17:
@flyingvince Alright. Case closed 🕵️

closed 2020-12-17 by swamper123